merge/skulls #109: [Doc] Backup Bios

ggg27, 2019-10-13:
**Question/Documentation**: Is it possible to back up the factory-default BIOS?

Or is it listed somewhere in case users ever want to switch back?

_____________________

**Note**: Some OSes like FreeBSD, NetBSD, Minix, e.c.t do not yet fully support Coreboot.

asvc, 2019-10-13:
@ggg27 , you can dump it with something like `flashrom -r original.bin`.

jcholsap, 2019-10-13:
Yes. Try it in circuit by making three copies and comparing the hashes. If the hashes don't compare then you may have power concerns. Some Lenovo mobos draw too much current and I have to pull the EEPROM  from the circuit to even read it.

ggg27, 2019-10-13:
@asvc @jcholsap For some reason I missed this when reading the manual (haven't started flashing yet): `sudo ./external_install_bottom.sh -m -k <backup-file-to-create>`

Does that backup the original BIOS? What does that command do?

______________________________________

> @ggg27 , you can dump it with something like `flashrom -r original.bin`.

Alright, btw do any of you know how large the BIOS backup should be?
Also, would I need to tell it anything about the ports (haven't used `flashrom`)?

asvc, 2019-10-13:
@ggg27 , 4mb top and 8mb bottom one. Please read the `man flashrom`.

ggg27, 2019-10-13:
@asvc What you appear to be communicating about is the pin layout.

I was curious if I had to enter the layout to flashrom or if this was by default.

>  Please read the man flashrom.

Will do, thanks!

closed 2019-10-13 by ggg27