merge/skulls #114: Please install dmidecode and run as root.

ggg27, 2019-10-19:
**Log**:
root@debian:/home/usr/coreboot/skulls-x230-0.1.6# apt install dmidecode
Reading package lists... Done
Building dependency tree       
Reading state information... Done
dmidecode is already the newest version (3.2-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@debian:/home/usr/coreboot/skulls-x230-0.1.6# ./x230_skulls.sh
Please install dmidecode and run as root.
root@debian:/home/usr/coreboot/skulls-x230-0.1.6

As you can see, I am under `su` and have `dmidecode`.

pahakalle, 2019-10-19:
Do you get the same error if you run `dmidecode -t bios | grep Vendor | cut -d':' -f2`

merge, 2019-11-27:
should be fixed by https://github.com/merge/skulls/commit/3ecfde0aea45b4ef69b19254b5c781823d1071d0

closed 2019-11-27 by merge