merge/skulls #115: [Doc] How to Use E-Z-Hook XKM

CanFixIt3242, 2019-10-20:
**Doc/Question**: How to use [E-Z-Hook](http://catalog.e-z-hook.com/viewitems/test-hooks/e-z-micro-hooks-single-hook-style) [XKM](https://catalog.e-z-hook.com/item/test-hooks/e-z-micro-hooks-single-hook-style/xkm-1)?

I purchased one XKM Micro-Hook double hook to Coreboot my x230 ThinkPad. 
However, I realized there is basically no documentation for how to use it.

The "SOIC Clip" has 8 pins while the XKM seems to only have 2. 
But, the Clip uses 6 female jumper wires.

**TL**;**DR**: Is there any documentation on how to use the E-Z-Hook](http://catalog.e-z-hook.com/viewitems/test-hooks/e-z-micro-hooks-single-hook-style) [XKM](https://catalog.e-z-hook.com/item/test-hooks/e-z-micro-hooks-single-hook-style/xkm-1)?
How many hooks is it required to flash?

jcholsap, 2019-10-20:
You need pins 1,2,4,5,6,8.

jcholsap, 2019-10-20:
i recommend this one:
https://www.amazon.com/CPT-063-Test-Clip-SOIC8-Pomona/dp/B00HHH65T4

jcholsap, 2019-10-20:
[image]

jcholsap, 2019-10-20:
You have the 8-pin SOP. Everyone reports that you can leave pins #3 and #7 not-connected.

jcholsap, 2019-10-20:
> You need pins 1,2,4,5,6,8.

Whatever programmer you use, you must use all these pins to correctly negotiate the SPI interface for both reading and writing.

jcholsap, 2019-10-20:
[image]

If you are using RPI, you can see that GPIO pins 17,19,21,23,25 and 24 are the ones you want.

jcholsap, 2019-10-20:
Steps 1, 2 and 3 give a pretty good explanation of what is going on, if that interests you any.

https://www.instructables.com/id/How-to-Design-with-Discrete-SPI-Flash-Memory/

CanFixIt3242, 2019-10-20:
@jcholsap Ok, so for future reference users would need [E-Z-Hook](http://catalog.e-z-hook.com/viewitems/test-hooks/e-z-micro-hooks-single-hook-style) [XKM](https://catalog.e-z-hook.com/item/test-hooks/e-z-micro-hooks-single-hook-style/xkm-1) [hooks 3x](https://cart.thomasnet-navigator.com/cbcheckout/addtocart?type=1&returnurl=https%3a%2f%2fcatalog.e-z-hook.com%2fitem%2ftest-hooks%2fe-z-micro-hooks-single-hook-style%2fxkm-1&Token=DaFiX8KNVP7I5FBvkPCIeAxh_MWa0T_WIZS2kRL3ABKWnkNReyvc1-XlQr9-cb9QMKNOdhVaV06m8mK_2W4p2g%2c%2c&recordIds=60658&selectedUOM=1)?

However, you still appear to need 6 jump cables, how do those plug into the hooks?
Perhaps, I have it reverse and the tip fits into the chip and the sheaved talon-like things are for the cables?

**Example Image of Hook**:

[image]

_______________________

> i recommend this one:
> https://www.amazon.com/CPT-063-Test-Clip-SOIC8-Pomona/dp/B00HHH65T4

It seems like there is no way to flash the chip without multiple hooks, so I will probably use this as it seems to be more well documented. 

Thanks you for your help!

____________________

jcholsap, 2019-10-20:
@CanFixIt3242 You're welcome. A few pointers:
1. This process requires diligence. Human error is very common.
2. Buy the Pomona brand. You can find cheap Chinese SOIC8 clips. But *many* people have complained about defective cheap clips causing intermittent faults that wastes hours of valuable time.
3. The worse mistake you can make is to rotate your clip 180 degrees. This will put your programmer Vcc to the chip's #4 Ground pin and your programmer Grounnd pin to the chip's #8 Vcc pin. This is called reverse bias and will literally melt your EEPROM. Otherwise these older SPI EEPROM's are hardy and forgiving of errant signals as long as they are within the voltage tolerance.
4. As a policy (and since you are new at this): Remove all - every possible - power source, including every kind of battery, from the motherboard. Power on your programmer last. Because the other worse thing you could do is accidentally reverse bias or short circuit something on the motherboard.
5. Expect that there will be a mistake (see #1 above).

CanFixIt3242, 2019-10-22:
> The worse mistake you can make is to rotate your clip 180 degrees. This will put your programmer Vcc to the chip's pin-4 Ground pin and your programmer Ground pin to the chip's pin-8 Vcc pin. This is called reverse bias and will literally melt your EEPROM. Otherwise these older SPI EEPROM's are hardy and forgiving of errant signals as long as they are within the voltage tolerance.

Does the clip even attach if you rotate it 180 degrees?

_________________________________

> Expect that there will be a mistake (see step 1 above).

What type of mistakes should I be concerned about? 
This is my live personal laptop, I have backup machines but I cannot really destroy this one.

Obviously, I plan to backup the original BIOS.

jcholsap, 2019-10-22:
If you get to the place you can read your EEPROM, then you've overcome the most probable mistakes. Reversing DO and DI is easy to do. I get pins 5 and 6 confused, but that's probably just my dysnomia. Clip moves, loosens; that easily happens. Forgot to remove the laptop or CMOS battery. Bad/flakey jumper wire/connection. Got the wrong hole on the breadboard. Lots of silly simple stuff. Other than the 180 degree (clock face perspective) thing -- Disclaimer: assuming you're of sound mind and reasonable intelligence -- you'll be fine.

CanFixIt3242, 2019-10-24:
> If you get to the place you can read your EEPROM, then you've overcome the most probable mistakes. Reversing DO and DI is easy to do. I get pins 5 and 6 confused, but that's probably just my [...]

I am still really confused about your warning on not turning it 180 degrees.

Obviously, you cannot move it 90 degrees and moving the pin 180 degrees looks identical.
Are you stating to making sure you wire it properly left-right (or right to left)?

jcholsap, 2019-10-24:
Let's say you match everything up correctly on your clip and to your programmer. Pin1 to pin1, check. Pin2 to pin2, check... All's is well. Then whilst moving about, oops, you bump the clip off. No worries. You just clip it back on. But wait. It fits two different ways (0 degrees and 180 degrees). It isn't keyed. So you could rotate your clip 180 degrees and reverse your pins 4 and 8. Okay?

Thrilleratplay, 2019-10-24:
@CanFixIt3242 There is a small circular pit in one of the corners of the 8-pin SOIC chip to indicate "pin 1", it is in the diagram above, the image that was recently merged and [an SVG here](https://gist.github.com/Thrilleratplay/60dff7cf849de5bdb84d91cabbd4fcfd)