merge/skulls #117: Heads and ME Neutralize compatible?

0xb100d, 2019-10-23:
I looked and did not see this elsewhere, sorry if it is a repeat.

1) Does Heads still work if we run the ME_cleaner?

2) Also is this compatible with TPMv2? 

3) If I just do ME cleaning and coreboot, can I later install Heads via software only (I believe I read this), or do I have to physically flash it at the same time?

Thank you so much!

CanFixIt3242, 2019-10-24:
It is a bit hard to find, but the docs to clearly state how to disable Intel ME:

- https://github.com/merge/skulls/blob/master/x230/README.md#ifd-unlock-and-me_cleaner-the-8mb-chip

I am unaware of any issues with running `me_cleaner` before before starting to install Skulls.
However, I am still working on installing Skulls on my PC for the first time.

_________________________________

**Note**: I assume you will be using Skulls on the x230 ThinkPad, as nothing else is officially supproted.

CanFixIt3242, 2019-10-24:
@0xb100d 

- [Moving to Heads](https://github.com/merge/skulls/blob/master/x230/README.md#moving-to-heads) from Skulls

>     3\. If I just do ME cleaning and coreboot, can I later install Heads via software only (I believe I read this), or do I have to physically flash it at the same time?

______________________________________

Hopefully, somebody else can answer the rest of the questions you have posted.

merge, 2019-10-24:
> I looked and did not see this elsewhere, sorry if it is a repeat.
> 
>     1. Does Heads still work if we run the ME_cleaner?

It does on the x230.

> 
>     2. Also is this compatible with TPMv2?

why do you ask? on the x230 we don't have TPMv2 hardware. Coreboot supports TPMv2 too.

> 
>     3. If I just do ME cleaning and coreboot, can I later install Heads via software only (I believe I read this), or do I have to physically flash it at the same time?

if you have IFD unlocked during external flashing of Skulls (for example), you can flash Heads via software.

> 
> 
> Thank you so much!

0xb100d, 2019-10-27:
> > 2. Also is this compatible with TPMv2?
> > ```
> 
> why do you ask? on the x230 we don't have TPMv2 hardware. Coreboot supports TPMv2 too.

I thought I read that the x230 can be upgraded to TPMv2? Very disappointing if it can't be done. 

https://pcsupport.lenovo.com/ca/en/downloads/ds032441
https://thinkdeploy.blogspot.com/2018/06/upgrading-tpm-spec-12-to-20-on-thinkpad.html

It is a hardware constraint?

jcholsap, 2019-10-27:
My understanding is that TPM v2.0 adds a hash check to firmware blocks that the EC verifies for "security". The result probably being, since mods change the hash value, that the EC would do whatever it does with an invalid hash. Maybe ME cleaner would remove this "feature" from the EC? Maybe installing TPM v2.0 starts with taking a hash of your current firmware state? Lots of variables. I'm curious to know what it'll do to you mods if you try it. I'm wondering if the update changes code within the EC.

0xb100d, 2019-10-27:
@jcholsap hopefully someone has the courage to test. TPM v2 seems substantially different and more advanced than v1

0xb100d, 2019-11-15:
@merge are you sure you the x230 cannot be updated tpmv2?

ami7az, 2019-12-11:
I found this by chance, if it's of any help:
> [SeaBIOS 1.10.0](https://seabios.org/index.php?title=Releases#SeaBIOS_1.10.0)
> Available on 20161026. Major changes in this release:
> Initial support for Trusted Platform Module (TPM) version 2.0 
So it should be at least in the skulls-firmware?