merge/skulls #118: [Doc] Cleanup Documentation for "8 Pin SOIC Clip" Pins

CanFixIt3242, 2019-10-24:
**The Issue**: The current docs don't give much information for the pins of the clip.
The x230 docs basically just states to set up the RPI and connect to laptop.

It doesn't even appear to state which chip to connect to, considering the x230 can have two.
[[Example](https://raw.githubusercontent.com/merge/skulls/master/x230/rpi_clip.jpg)]
____________________________________

**Potential Solution**: 

I crafted the following based on an image found here.

[image]

There appears to be 9 jumper cables, but the SOIC Clip only has 8.
<br/> How do I wire the clip?

______________________________

**Note**: Feel free to tell me to RTFM, but please tell me where the information is located,
____________________________________

jcholsap, 2019-10-24:
For simplicity, I say use GPIO pins 17, 19, 21, 23, 25 and 24. But you are free to choose whatever GND and/or 3.3V you like.
In your graphic, you just want the SPI signals for the first slave. (RPI can manage 2 slaves.) They are labled in your figure: VCC (17), MOSI (19), MISO (21), CLK (23), CS (24), GND (38). My suggestion is to use GND (25). But that's just a preference.
TX and RX have no relevance for this discussion.

jcholsap, 2019-10-24:
[image]

You have an 8-pin SOP

CanFixIt3242, 2019-10-24:
@jcholsap Sorry I I am being a bit too thorough on this.
I realize that you also answered many of my other questions. 

> In your graphic, you just want the SPI signals for the first slave. (RPI can manage 2 slaves.)

Okay, so use the chip furthest away from the screen.

> For simplicity, I say use GPIO pins 17, 19, 21, 23, 25 and 24. But you are free to choose whatever GND and/or 3.3V you like.

Ok, so basically do:

[image]

> pin 5 is the same as MOSI
pin 2 is the same as MISO
THe black dot on the chip package corner indicates pin 1. Usually the mobo also has a triangle marking at the pin 1 position. You want to be really very careful and make sure you find pin 1.

Oh, ok. So you want to find the first pin in order to find every other pin.
Is it not always located in the upper left?

____________________________

>  In your graphic, you just want the SPI signals for the first slave. (RPI can manage 2 slaves.) They are labled in your figure: VCC (17), MOSI (19), MISO (21), CLK (23), CS (24), GND (38). My suggestion is to use GND (25). But that's just a preference.

I am honestly lost of what you mean here.

jcholsap, 2019-10-24:
This might help you some:
https://pinout.xyz/pinout/#

jcholsap, 2019-10-24:
Your 8-pin SOIC diagram is... unusual. But yes the convention is to put the "dot" in the upper left corner. Then count pins counter clockwise from there.

jcholsap, 2019-10-24:
Raspberry Pi                           EEPROM
GPIO SPI pins:                        SOIC8 pins: 
==================================
VCC (17),                                VCC (8)
MOSI (19),                               SI (5)
MISO (21),                               SO (2)
CLK (23),                                 CLK (6)
CS (24),                                   CS (1)
GND (25).                                GND (4)

CanFixIt3242, 2019-10-25:
@merge Alright so the actual pin out is (considering most users will probably have a screen and keyboard):
 ```
Edge of pi (furthest from you)
   L                                               CS
   E                                                |
   F +---------------------------------------------------------------------------------+
   T |  x   x   x   x   x   x   x   x   x   x   x   x   x   x   x   x   x   x   x   x  |
     |  x   x   x   x   x   x   x   x   x   x   x   x   x   x   x   x   x   x   x   x  |
   E +----------------------------------^---^---^---^-------------------------------^--+
   D                                    |   |   |   |                               |
   G                                   3.3V MOSIMISO|                              GND
   E                                 (VCC)         CLK
     Body of Pi (closest to you)

```

However, this still does not fully answer the question of which pins on the RPI to connect to the pins on the clip.

jcholsap, 2019-10-25:
Pin matchups are as follows:
RPI GPIO -> EEPROM pins:
 VCC (17) -> VCC (8)
 MOSI (19) -> SI (5)
 MISO (21) -> SO (2)
 CLK (23) -> CLK (6)
 CS (24) -> CS (1)
 GND (25) -> GND (4)

CanFixIt3242, 2019-10-26:
> Pin matchups are as follows:
> RPI GPIO -> EEPROM pins:
> VCC (17) -> VCC (8)
> MOSI (19) -> SI (5)
> MISO (21) -> SO (2)
> CLK (23) -> CLK (6)
> CS (24) -> CS (1)
> GND (25) -> GND (4)

Alright so, although i would love to learn, I don't know what `VCC`, `MOSI`, `SI`, `SO`, `CLK`, `CS`, or `GND` mean.

However, I can put together that pin 17 goes to pin 7 on the clip.

________________________________

@jcholsap Thanks