merge/skulls #121: ifdmegbe gigabit ethernet blob? + including iPXE?

ami7az, 2019-11-13:
Does Skulls keep the gigabit ethernet blob of the bottom ROM (ifdmegbe)?
Or is it being replaced or put dormant, with something entirely open source instead?
In case of the latter, would it be possible to also include iPXE for free ethernet (not Wi-Fi!) connections as a payload, or would this lead to more attack surface?
Thank you!

merge, 2019-11-21:
our installation keeps your gbe blob (and ifd) untouched and only replaces the bios flash region.

ami7az, 2019-12-11:
Thanks! It would be great if this could be replaced with something FOSS in the future, though.

Once the linux kernel boots, is it FOSS only then by the drivers, and inactive?
Or is this blob always in charge?

(I'd like to have the least amount of blobs possible.)

Th3Fanbus, 2020-05-29:
The GbE region is just data. If that scares you, maybe you should consider using [something that truly respects your freedom](https://en.wikipedia.org/wiki/Typewriter).

Yes, unironically.

Thrilleratplay, 2020-09-07:
Anyone who would like to generate their own [gbe can with this bincfg set and spec](https://review.coreboot.org/c/coreboot/+/44510).  I strongly suggest modifying it and using your own MAC address.  IBoot and PXE are disabled.

@Th3Fanbus That isn't necessarily true.  When reading through the data sheet, I saw where the IBoot Advanced Power Management does interact with Management Engine (ME).  I have no idea what the possibilities are or if there is the ability to exploit this, but there is more than just data in certain parts.