merge/skulls #122: x230 hardware switch for wireless networking does not work?

0xb100d, 2019-11-21:
Could not find this brought up elsewhere.

Totally baffling to me, as it means that the hardware switch is not a hardware switch at all apparently?

Wifi state on Qubes 4.1 does not change at all when the wiresless networking switch is in either position. Only thing I can think of is that I have a 3band atheros wifi card, with the red wire for the cellular card plugged into it. I have not compared the wifi speeds for hardware switch on/off positions. I would assume though that (especially) the cellular card would also turn off with the hardware switch.

Side question, does anyone know which of the other two antenna cables is preferable to use when installing a 3x3 wifi card in the x230? There was a blue a grey a black and a red one. I believe the normal wifi card takes the black and grey ones, and if I recall I plugged the red one into the third antenna spot.

Thank you for your time.

jcholsap, 2019-11-21:
I have a T430s, which I believe is very similar. The "WiFi" switch on the T430s is right where the bluetooth module is located. So, maybe the switch is only for BT?
The T430s WiFi antenna colors are Gray, Black and White. The WiMaxx antennas are Blue and Orange. But I don't know if the other antennas are designed for 2.45GHz or not. I'll be taking mine apart soon and I'll look at the antennas and let you know what I find. Also you can try signal strength with and without the other antenna. I know on the 4th gen Lenovo laptops teh antennas are different.

jcholsap, 2019-11-21:
RE: Wireless switch: Looking at the X230 HMM (hardware maintenance manual) and testing on my unmodified T430s, the switch turns "wireless" ON/OFF. Also, noticing that FN F5 also turns "wireless" ON/OFF, it looks like the controller (EC) is involved with this feature. If you've modified the BIOS, i.e., removed the WiFi whitelist, then most likely the EC is ignoring this "feature", and it is problaby not a function of Qubes. I believe the similar issues pertain to the EC with removing the battery whitelist.

jcholsap, 2019-11-21:
RE: Antennas: The WWAN antennas are about three times larger than the WiFi antennas. The Orange WWAN antenna is marked as "Main". This may just mean that it has more gain. If that's the case then using the WWAN antennas for your WiFi card would be preferable. Or it could mean that it's tuned to a different bandwidth. In that case, they won't likely help your WiFi card unless there's some overlap on the antenna bandwidths. You've got me curious about it though because of the possible much greater gain with the WWAN antennas. Pics attached for you.
[image]
[image]

jcholsap, 2019-11-22:
RE: Antennas: I came across one of these with an AirPrime 7750 WLAN card installed. Did some research and found that the antennas are just for cellular networks. But the lower limit is 2500 MHz which is almost at the upper end of the WiFi bandwidths. So, just maybe, you could pull in that 14th channel with a WLAN antenna. But probably not and even so, it's just one channel.
[image]
[image]
[image]