merge/skulls #126: build.sh error: defconfig no file or directory

tjburrows, 2019-12-05:
I am trying to build skulls 0.1.7 from the release source code zip by running build.sh.  However, I get the following error.  I am using Debian 10.  Thanks for the help!

> travis@debian:~/Downloads/skulls/skulls-0.1.7$ ./build.sh x230
> Beginning download of coreboot-4.11.tar.xz...
> Downloading coreboot-4.11.tar.xz complete
> cp: cannot stat '/home/coreboot/scripts/defconfig': No such file or directory
> Using default config
> *** Default configuration is based on 'configs/defconfig'
>\***
> *** Can't find default configuration "configs/defconfig"!
> \***
> make: *** [/home/coreboot/cb_build/util/kconfig/Makefile:99: defconfig] Error 1
> Error: Expected config file (/home/coreboot/cb_build/.config) not present.
> Please specify a config file or run 'make menuconfig' to
> generate a new config file.
> make: *** [Makefile:156: real-all] Error 1
> Uh oh. Things did not go according to plan.

merge, 2019-12-05:
please use `build.sh` inside of the x230 directory: `cd x230` `./build.sh`

tjburrows, 2019-12-05:
Oh, thanks!

closed 2019-12-05 by tjburrows