merge/skulls #127: RAM issue - not all ram is usable

PietroProraso, 2019-12-06:
Hi,

I am using skulls on my x230t. Thanks for providing this service!

However, the system recognizes only (roughly) 8GB of the installed 16GB RAM.

The memory I am using is "HyperX Impact", two 8GB modules. These modules have an automatic overclocking functionality, i.e. the modules are supposed to run with the maximum speed the system can provide.
I guess this could be part of the problem.

Is there any way I can use all of my RAM? Thanks for any help!

jcholsap, 2019-12-07:
@PietroProraso  
My understanding of the HyperX modules  -- BTW, I've always wanted those -- is that they can operate up to a frequency of 1066MHz. Your motherboard supports something from 800MHz and up, but you don't know what the maximum, stable memory bus frequency is. My understanding is that your HyperX XMP tables are telling coreboot raminit that it can do 1066MHz but your motherboard probably isn't stable at that frequency.
So, I think the question becomes: 
1. What is the maximum, stable frequency for your particular motherboard's memory lanes?
2. Are you willing to experiment to find the optimal frequency? (It's somewhere between 800-1066MHz)

I think you are going to need to roll up your sleeves and try different settings for max_mem_clock_mhz. This is something I plan on doing but have not personally tried yet. But hopefully my research can help you.
[image]

PietroProraso, 2019-12-07:
@jcholsap 

Hi,
thanks for your answer.

I'm willing to go through different frequency settings. My problem is, I don't have experience compiling coreboot, i.e. I can compile coreboot successfully, but i don't know what at least half the settings mean, that I can choose with `make nconfig`.

Could you maybe provide a config file that I can load, which is similar to what is used for the skulls coreboot image?

jcholsap, 2019-12-07:
@PietroProraso 
Not at this time, but it  is on my gantt chart. Think of the possibilities. You can also overclock your CPU. Definitely worth figuring out.

jcholsap, 2019-12-07:
Just compile for x64. You don't need to crosscompile anything.

kamilr, 2019-12-10:
My friend upgrade my RAM to 16GB (HyperX DDR3L 1600 12800 cl9) from 8. I still have original (2.66) Lenovo bios, which shows on init also 8GB but the OSes see 16. Also in Lenovo Diagnostics (after Boot menu) there are also 16GB. The same in case of QubesOS, it's 16 already. Today I  wonna reflash my x230t motherboard with Skulls. Will see, and let You know here.

PietroProraso, 2019-12-10:
All 16GB RAM are usable now.

I used [Thrilleratplay/coreboot-builder-scripts](https://github.com/Thrilleratplay/coreboot-builder-scripts) (which, according to the skulls README are used by skulls, too) to build a coreboot rom (I built the coreboot release version 4.11).

With this coreboot rom, all 16GB are detected, regardless of whether I have modified `max_mem_clock_mhz` comparable to [this commit in coreboot](https://github.com/coreboot/coreboot/commit/7bddd30e948e6c3856336e77221fdbe5ba749453#diff-46608db41ce5be0ddd02dd64d54be77a) to values of 666 or 1066. 
I am now using the unmodified coreboot release version 4.11, built with Thrilleratplay's coreboot-builder-scripts and all my RAM is used.

Some change in coreboot from release version 4.11 to the version used for skulls-x230 0.1.7 (which according by the release date of skulls 0.1.7 must be < 4.11) must be the cause.

In any case: I'm happy with my installation now :).

jcholsap, 2019-12-10:
@PietroProraso Very Cool!!! Well done. Personally, I would also stress (memtest) the system clocking the memory lanes at that frequency. I'm fairly sure 1066MHz is at the maximum. Since it's Lenovo, I'd expect design quality and really wouldn't be so concerned. But I'm a bit zealous about stability/reliability. :-)

ami7az, 2019-12-11:
What kind of RAM is best to get?
I've read of people buying very expensive DDR3 SO-DIMM DDR3L-2133 (PC3L-17000) 2133 MHz/1.35 V memory because apparently the x230/t's mainboards support it, but the official Product Specifications Reference (PSREF) only lists DDR3-1600: "16GB max / PC3-12800 1600MHz DDR3, non-parity, dual-channel capable, two 204-pin SO-DIMM sockets" (1.50 V). And the Hardware Maintenance Manual (HMM) doesn't seem to mention the specifics at all.

Where did you all get your knowledge?

And does the lower power consumption / higher clockspeed actually make a noteable difference in speed / battery life?
Could you please post some benchmarks?

PietroProraso, 2019-12-11:
The RAM I use (see above) is running at 800MHz, corresponding to DDR3-1600 speeds, see here:

```
$ sudo dmidecode --type 17
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.

Handle 0x000A, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x0000
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 8192 MB
	Form Factor: SODIMM
	Set: None
	Locator: Channel-0-DIMM-0
	Bank Locator: BANK 0
	Type: DDR3
	Type Detail: Synchronous
	Speed: 800 MT/s
	Manufacturer: Kingston
	Serial Number: 4a12761b
	Asset Tag: Not Specified
	Part Number: KHX1600C9S3L/8G
	Rank: 2
	Configured Memory Speed: 800 MT/s
	Minimum Voltage: Unknown
	Maximum Voltage: Unknown
	Configured Voltage: Unknown

Handle 0x000B, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x0000
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 8192 MB
	Form Factor: SODIMM
	Set: None
	Locator: Channel-1-DIMM-0
	Bank Locator: BANK 0
	Type: DDR3
	Type Detail: Synchronous
	Speed: 800 MT/s
	Manufacturer: Kingston
	Serial Number: 4b122c11
	Asset Tag: Not Specified
	Part Number: KHX1600C9S3L/8G
	Rank: 2
	Configured Memory Speed: 800 MT/s
	Minimum Voltage: Unknown
	Maximum Voltage: Unknown
	Configured Voltage: Unknown

```

Because of this, I guess, this is the maximum supported by my x230t. 

Can't say anything substantial to your other questions, though.

jcholsap, 2019-12-11:
KHX1600C9S3L/8G is an 800MHz DDR3 module. Versus the HyperX Impact HX316LS9IBK2 modules which are 1066MHz. And here is another 1066MHz SODIMM: G.SKILL Ripjaws F3-2133C11D. 

I believe that the 1066MHz modules *are* supported by design, but that's the maximum, unless you OC. 

Personally I have 933MHz (PC3-14900) modules because the others are so hard to find or very expensive. Here's a performance comparison: https://imgur.com/a/rH7vU

jcholsap, 2019-12-11:
@ami7az Lenovo doesn't keep PSREF (or the hmm) updated.

PietroProraso, 2019-12-14:
@jcholsap 

> Just compile for x64. You don't need to crosscompile anything.

This is wrong.

I have written above that I had to use coreboot-builder-scripts to get a sucessfull build. My own coreboot builds resulted in an unbootable device (black screen after pressing the power button).
I was able to recover my machine only by externally re-flashing the skulls image.

@jcholsap , please do not give advice if you don't know what you are talking about! If you are guessing, state that you are guessing.
This one sentence of yours has cost me a whole day of my life, it could have rendered 200 Euros of perfectly fine hardware useless. And potentially other people will read this, too, and experience the same trouble.

I'll edit my opening post to make clear that cross compiling is necessary. As far as I understand, cross compilation here is not necessary because the system architecture (obviously), but using the cross compiling toolchain ensures that the resulting image has the very same structure. Building with the default toolchain will typically not result in the very same image structure as shown here: https://doc.coreboot.org/mainboard/lenovo/xx30_series.html

jcholsap, 2019-12-15:
@PietroProraso My apologies for my careless reply and any inconvenience it may have caused you. You remind me of the importance for being diligent in making posts here. As you have pointed out, the instructions are quite clear about how to build the coreboot toolchain.

ami7az, 2019-12-15:
Thanks @PietroProraso! & @jcholsap!
So, [2 x 800MHz = 1600MHz?](https://en.wikipedia.org/wiki/DDR3_SDRAM#Modules) And Transfer rate (MT/s) = 2 x 1600?
Or why PSREF's "PC3-12800 1600MHz DDR3"?

jcholsap, 2019-12-16:
@ami7az 
> So, [2 x 800MHz = 1600MHz?](https://en.wikipedia.org/wiki/DDR3_SDRAM#Modules) 

Yes, data is transferred on both the rising and falling edges of the clock signal, doubling the effective rate.

> And Transfer rate (MT/s) = 2 x 1600?

I don't remember the throughput off hand. Just look it up.

>  Or why PSREF's "PC3-12800 1600MHz DDR3"? [sic]

My observation has been that Lenovo just doesn't bother to keep their docs updated. So along comes something *new, like 1066MHz SODIMMs, and Lenovo doesn't even acknowledge it. 
*new -- as in after the docs were written.

If you go for the PC3-17000 modules, *please* share your experience with us!

PietroProraso, 2020-01-03:
With skulls-x230 0.1.8 the RAM issue because of which I started this thread is resolved. 

I'm closing this issue.

closed 2020-01-03 by PietroProraso