merge/skulls #128: CBFS_SIZE needs to be adjusted to give BIOS region more space

tlaurion, 2019-12-08:
https://github.com/merge/skulls/blob/1cde00d5c9c66e14c06ae1f850efa27bd4586052/x230/free-defconfig-4357a820764#L4

@merge: Not sure how to calculate this value, unfortunately?

jcholsap, 2019-12-08:
This might help... excerpted from https://www.coreboot.org/Board:lenovo/x230
Flashing
Hardware Flashing

X230 has 2 flash chips of 8M and 4M. They're concatenated to one virtual flash chip of 12M (the 8M chip is the first portion of this virtual chip, and the 4M is the final portion) which is itself subdivided in roughly in 3 parts:

    Descriptor (12K)
    ME firmware (5M-12K)
    System flash (7M)

ME firmware is not readable. Vendor firmware locks the flash and so you need to flash externally.

Proceeds as follows:
    Connect your external SPI flasher to the top SPI chip which is under palm resting space, on left side of the board. It's a 4M chip. IF you've chosen CBFS_SIZE 4M or smaller that' the only chip you need to reflash.

Write the flash. Since you have to write only top 4M, first split out those 4M:
     dd of=top.rom bs=1M if=build/coreboot.rom skip=8

Use flashrom to flash top.rom.

merge, 2019-12-09:
I'm not sure what you mean here. We use 4M CBFS size in Skulls, so that flashing the 8M chips remains optional and you have a working system after only flashing the 4M chip.

tlaurion, 2019-12-09:
That was aimed at heads for your insight on how to calculate bios maximum BIOS region that can be given when generating the full 12mb rom, with ifd and neutered+deativated ME.

Misplaced. Closing

closed 2019-12-09 by tlaurion