merge/skulls #130: Flashing back to the stock x230 BIOS

stealth399, 2019-12-14:
Quick question -- if I flash back to the original bios via booting from the lenovo update image, will it re-lock my flash making the SPI programmer necessary again?

Should I instead use ifdtool to unlock my original bios backup, then use flashrom to reinstall the modified backup if I want the original bios, thus allowing me to reinstall coreboot without SPI in the future?

Apologies if this doesn't make sense. Thank you for your work!

jcholsap, 2019-12-14:
Good question and it does make sense. I don't know the answer but have a couple of concerns to share:
* Make sure the Lenovo image is compatible with your EC version, and
* If you use the "security featured" image that was released this year, then you will not be able to modify the BIOS and may not be able to revert to coreboot by *any* documented means.

tlaurion, 2019-12-14:
On December 14, 2019 5:07:37 PM UTC, Jay Holsapple <notifications@github.com> wrote:
>Good question and it does make sense. I don't know the answer but have
>a couple of concerns to share:
>* Make sure the Lenovo image is compatible with your EC version, and

False:
>* If you use the "security featured" image that was released this year,
>then you will not be able to modify the BIOS and may not be able to
>revert to coreboot by *any* documented means.

You won't be able to flash unsigned EC, stopping you from using thinkpad-ec GitHub project to use x220 keyboard and non original batteries patches.

-- Sent from /e/ Mail

mszumilak, 2019-12-14:
I cannot agree. I flashed my bios to the newest one (protected), then I realized that I should apply patches. Downgrading original bios allowed to me downgrade EC as well and I installed patched version. Sadly I didn't enable patch for battery and left the default patch for keyboard which messed up couple of keys. I have to flash back from coreboot using backup images, install properly patched EC, then flash to coreboot once again.

jcholsap, 2019-12-14:
@mszumilak 

> Downgrading original bios allowed to me downgrade EC as well and I installed patched EC version.

Thanks for the correction. That's really good news. I was unaware that anyone had tried this.

stealth399, 2019-12-15:
Attempted to flash back to a me_cleaned and unlocked stock bios from the backup. I see the lenovo startup screen, but I get a 0271 Real Time clock error. I can then UEFI boot to a flash drive, but flashrom always fails stating that it can't get permissions to flash. I think I need to disable secure boot, but I can't as the laptop now hangs when trying to enter the bios.

Not sure what to try next. Guessing I'll open it back up and just flash the top chip with coreboot again to see if that gets it booting.

I'm wondering if the bios needs ME to boot? I tried clearing CMOS thinking that would help, but no dice. I can't get the BIOS menu to load.

Bizarre.

stealth399, 2019-12-15:
Looks like my live CD may enable something called kernel lockdown mode, so hopefully if I disable that and do the iommu=relaxed trick, I'll at least be able to get coreboot booting again.

jcholsap, 2019-12-15:
@mszumilak 
> and I installed patched EC version.
Are you saying that you are using the latest EC firmware? At first I thought you had just reverted to the older version. I'm interested in doing this but feel concerned that the updated EC could be handling the BIOS differently. But if you have it working, and are able to rollback, then it's something I'd like to try.

stealth399, 2019-12-15:
Tried to avoid re-attaching leads, so I went ahead and tried to completely restore the 12MB via an internal flash. Worked about as well as you'd expect -- have a nice brick now! Opening it back up and will restore contents to default to see if the aforementioned internal flash caused the BIOS error. If I still can't access the bios with my old backups, I'll try flashing from the USB boot CD. If that still doesn't work, I'll try coreboot once more. And if that doesn't work, I suppose I need a new motherboard. Doh!

stealth399, 2019-12-15:
Sorry for the massive number of comments here, but perhaps this will be useful for someone else someday. Restored both chips via flashing with spispeed 128 (read that online elsewhere, no idea if it mattered). I can get into BIOS again.

I'm wondering if me_cleaner is causing some sort of issue. I'm going to reflash the last unlocked EC bios, reset BIOS to defaults, disable secure boot, then do the skulls install again.

mszumilak, 2019-12-15:
@jcholsap 
> > and I installed patched EC version.
 
> Are you saying that you are using the latest EC firmware? At first I thought you had just reverted to the older version. I'm interested in doing this but feel concerned that the updated EC could be [...]

No, I'm EC version 1.13 pached using https://github.com/hamishcoleman/thinkpad-ec/

stealth399, 2019-12-15:
To finally end this saga, no, I do not believe what I said originally is possible. Manually flashed back to stock, reinstalled skulls, flashed back to lenovo bios via internal flash a second time, same real time clock error / fail to enter into the bios settings menu.

Because I disabled secure boot/UEFI I can still boot my OS though. Trying to flash coreboot back internally does NOT work for some strange reason after going back to the Lenovo bios -- guessing it's locking something?

'''
sudo flashrom -p internal --layout x230-layout.txt --image bios --write complete_sea.bin.new 
[sudo] password for work: 
Sorry, try again.
[sudo] password for work: 
flashrom v1.1 on Linux 5.3.15-300.fc31.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using region: "bios".
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found chipset "Intel QM77".
Enabling flash write... Warning: BIOS region SMM protection is enabled!
Warning: Setting Bios Control at 0xdc from 0x2a to 0x09 failed.
New value is 0x2a.
SPI Configuration is locked down.
PR1: Warning: 0x00b40000-0x00bfffff is read-only.
PR2: Warning: 0x00b10000-0x00b10fff is read-only.
PR3: Warning: 0x00ad0000-0x00adefff is read-only.
PR4: Warning: 0x00800000-0x00aaffff is read-only.
At least some flash regions are write protected. For write operations,
you should use a flash layout and include only writable regions. See
manpage for more details.
Enabling hardware sequencing due to multiple flash chips detected.
PROBLEMS, continuing anyway
Found Programmer flash chip "Opaque flash chip" (12288 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading old flash chip contents... done.
Erasing and writing flash chip... Transaction error between offset 0x00800000 and 0x00800fff (= 0x00800000 + 4095)!
Reading current flash chip contents... done. Looking for another erase function.
Looking for another erase function.
Looking for another erase function.
Looking for another erase function.
Looking for another erase function.
Looking for another erase function.
Looking for another erase function.
No usable erase functions left.
FAILED!
Uh oh. Erase/write failed. Checking if anything has changed.
Reading current flash chip contents... done.
Good, writing to the flash chip apparently didn't do anything.
This means we have to add special support for your board, programmer or flash
chip. Please report this on IRC at chat.freenode.net (channel #flashrom) or
mail flashrom@flashrom.org, thanks!
-------------------------------------------------------------------------------
You may now reboot or simply leave the machine running.
'''

Anyway, I previously reflashed the EC with the keyboard fix. It's all working now, but if you ever want to change ECs and remain on coreboot, it looks like you'll have to hardware flash unfortunately.