merge/skulls #131: Weird issues with skulls 0.1.7

ShFil119, 2019-12-22:
I've installed skulls with rpi 3. Before installation I've updated to newest available stock bios. Used nonfree version with vga blob. Also flashed bottom with me_cleaner.

Ok, issues are:
1. lower battery life time, I think around ~30%. 
Maybe connected:
`[    5.917177] acpi_call: Cannot get handle: Error: AE_NOT_FOUND`
2. Much longer cold start than stock. (bios part)  I think it's 10 seconds. Reboot takes only 2,5 sec.
3. seaBios loses ssd after hardreset, aka stopping x230 with power button, to fix it I need to boot something else and reboot. Physical reinserting doesn't help.
4. Noisy output of mei_me:
```
[    6.021382] mei_me 0000:00:16.0: wait hw ready failed
[    6.021390] mei_me 0000:00:16.0: hw_start failed ret = -62
[    6.021411] mei_me 0000:00:16.0: H_RST is set = 0x80000015

```
 ```
[    8.154710] mei_me 0000:00:16.0: wait hw ready failed
[    8.154716] mei_me 0000:00:16.0: hw_start failed ret = -62
[    8.154736] mei_me 0000:00:16.0: H_RST is set = 0x80000015
[   10.288036] mei_me 0000:00:16.0: wait hw ready failed
[   10.288043] mei_me 0000:00:16.0: hw_start failed ret = -62
[   10.288055] mei_me 0000:00:16.0: reset: reached maximal consecutive resets: disabling the device
[   10.288058] mei_me 0000:00:16.0: reset failed ret = -19
[   10.288060] mei_me 0000:00:16.0: link layer initialization failed.
[   10.288063] mei_me 0000:00:16.0: init hw failure.
[   10.288618] mei_me 0000:00:16.0: initialization failed.
```

Thanks for help. I haven't tried to downgrade, coreboot newbie here.

ShFil119, 2020-01-03:
Still happens on 0.1.8

ShFil119, 2020-01-15:
I've done more checks:
- firefox sometimes crashes,
- memcheck86+ freezes (no errors),

Looks like memory issue, but before flashing skulls everything seemed fine. 

I've tried to reseat ram modules (2 x 4 GiB 1600MHz), what else can I check/do?

jcholsap, 2020-01-16:
I'm willing to bounce ideas back to you.
* Reading your error log, it looks like a hardware device has failed to initialize after three attempts. It mentions "link layer" which makes me think of an ethernet interface. But I'm trying to remember from writing kernel modules, maybe "link layer" could refer to a driver interface as well?
* Definitely, I'd start with an older stock BIOS. Lenovo has done something with their latest BIOS release.
* If you want to test RAM, you'll need to put it in a machine that you know works and run memtest there.

ShFil119, 2020-01-19:
Not sure, if it's it, but I've found this comment:
https://old.reddit.com/r/coreboot/comments/ep14fj/ram_voltage_important/fel2l2i/

I also use 1,35v RAM, do you think that it can cause problem?

kubik369, 2020-01-20:
Hi, that's my comment :)

I had similar problems as you (I will be looking more into it after exams at my university end in mid February).

My configuration is 2x matched Crucial 8GB DDR3L RAM, 250GB Samsung 850 SSD and a pretty basic i5-3320M X230.

When I flashed skulls 0.1.7 (update to newest BIOS at the time - like 2 weeks back, no me_cleaner), the laptop started hard crashing in Linux almost immediately (this corrupted my system, it took me a few hours to fix :( ). My boot times were approximately the same as yours, haven't timed it, but it sure as hell wasn't faster than stock, reboots are just as you mention, quite fast. I haven't noticed seabios missing any devices even after hard-crashing and the laptop being forcefully turned off. I have noticed noisy output similar to yours.

Memtest is a separate chapter in this. I also had problems with memtest, not really crashing, as the loader was still spinning, but the progress just froze approximately at the same place each time. I have found on a few websites that the included newest version of memtest has this problem and after I tried booting the older version (4.3.7), this one did not have any problems. 

After trying to reseat my RAM and reading about problems with low-voltage RAM, I gave it a last try and I have tried building newer coreboot from the link in the skulls README (which I provided in the reddit thread). The build has newer version of both seabios (1.12.1 -> 1.13.0 iirc) and coreboot (I can look it up if it is needed) and it has been running almost flawlessly in the last 2-3 weeks - I don't know whether that is due to seabios or coreboot updates. There are only small annoyances which I will be looking into such as the fan-curve being too aggressive, the weird ME output and I still have boot times which are slower than stock.

I hope that we can piece together what is happening here :)