merge/skulls #132: Keyboard Issue on 0.1.7 Release

zackmagic42, 2019-12-24:
I'm having an issue using the nonfree release in 0.1.7, I'm running Manjaro and everything was fine before the flash. My X230 booted up fine but when I try to use the function key to control screen brightness the brightness up doesn't do anything and down disables the trackpad. Also I can't get the keyboard backlight to work at all or the led up top near the webcam.

merge, 2020-01-02:
have you somehow changed your EC independently of flashing coreboot (Skulls) before?

zackmagic42, 2020-01-02:
My EC was updated before the install of coreboot however the keyboard layout was fine until the install. I found that the I can in fact change the keyboard backlight by pressing FN+Backspace instead of spacebar and I can change the display brightness but only lower it. The keyboard shortcut for display brightness down instead disables the trackpad and display brightness up does nothing. The delete key was also swapped to the home key.