merge/skulls #134: Tpacpi-bat

compicat, 2020-01-05:
First of all thanks for your great work. Because of i m recalibrsting batteries in my shop, i need to use tpacpi-bat. Can i kindly learn is this patch merged to skulls last release? If it is not can i add for my own build? Thansk in advance. https://review.coreboot.org/plugins/gitiles/coreboot/+/93d6ba0889d0247ac264172858ef57698c960464

apraile, 2020-01-05:
Yes, it's been implemented since **skulls 0.1.2**.
See #81

compicat, 2020-01-05:
> Yes, it's been implemented since **skulls 0.1.2**.
> See #81

Thank you. I dont know why but tpacpi-bat is inactive because of it was superseded by natacpi. Natacpi and tpacpi-bat working together at my whitelist removed lenovo bios. Because tpacpi-bat is not working, recalibration battery is not working.

compicat, 2020-01-16:
Just for the others, i ve solved this via ectool. Below command is recalibrating battery of X230 on skulls by discharging and fully charging.

./ectool -w 0xb4 -z 0x06