merge/skulls #136: Add 1vyrain to the list of flashing methods

aladar42, 2020-02-03:
1vyrain officially supports skulls as o'er their homepage. I have tried this method and it works just fine on am x230.

https://1vyra.in/

merge, 2020-02-04:
looks a bit adventurous, but sure :) If you have tested it, can you document it in a short paragraph and do a merge request?

Dongerrs, 2020-02-04:
Is it possible to flash this OVER Skulls to get a 'standard' (modded) BIOS? 

See my previous post/issue regarding recovery to standard BIOS to 'unbrick' a third party battery

aladar42, 2020-02-05:
@merge I unfortunately don't have any PC access for an indefinite period of time, so I can't write up anything. However, the instructions are all on the github page, the USB exploit itself only requires you to press enter, pretty much. The only thing I would add is that skulls BIOS needs to be wget-able to flash it, so it should be uploaded to Mediafire or something.

@Dongerrs yeah, that should work. A premodded Lenovo bios is already supplied as a payload in the image.

tlaurion, 2020-02-05:
Limits are that external flashing is still required if modifying IFD (from neutering ME and making that space available to CBFS), so documentation should include those [limitations](https://github.com/n4ru/1vyrain/issues/2).

- Skulls releases could be downloaded from github instead of MediaFire (which tends to disappear)

pgera, 2020-02-08:
Just some context. This is essentially a natural conclusion to https://github.com/merge/skulls/issues/49#issuecomment-462243384 . So overall, it's a great idea to lower the barrier of entry for coreboot.  @gch1p has also been doing some work in this space. https://review.coreboot.org/c/coreboot/+/37115 is of interest since it enables the soft temp disable mode for Intel ME. That is the best you can do right now without physically writing to the me region. The documentation around this was also recently merged (https://doc.coreboot.org/mainboard/lenovo/ivb_internal_flashing.html)

samveen, 2020-03-05:
@merge would it be possible to add the *.rom and the *.rom.sha256 files directly as release assets? I see that it will increase the project's storage requirements, but on the flip side, it's going to save people from link rot and things even worse on `MediaFire`.

merge, 2021-03-27:
the images are just compressed as a tarball. why would unpacking a tarball be a problem for users?

aladar42, 2021-03-27:
Because the flashing method only supports raw ROM files. The live USB doesn't have tar packaged.

-------- Original Message --------