merge/skulls #137: Revert to Original BIOS/ROM

Dongerrs, 2020-02-04:
Need to re-flash original bios, as I have a third party battery that has stopped charging due to me setting stupid thresholds (See the TLP FAQ). Want to see if i can remove/reset the threshold using original (EC Modified for 3rd Party Battery/Hamish Coleman moded) BIOS

Want to know easiest way to reflash the backups?

Presumably i can do it in software, as skulls is currently installed and i havent locked the bios down after installing but.... 

1. do i need to reflash the whole 12Mb of ROM - with a single 12mb backup?
2. do i need to reflash all 12Mb of ROM - with 2 separate 8Mb/4Mb Backups?
3. do i need to reflash the top ROM only?
4. do i need to reflash bottom ROM only?

As i have run ME_Cleaner too, will i actually get a working x230 or a brick? Have seen some reports of people trying to go back to standard BIOS and not being able to boot. Also seen some reports of people downloading standard ISO image from Lenovo and restoring that way..? If use standard image, is there a version that i should not go above (as have also read that later versions locked down the EC)?

Anyone with any ideas/experience..?

digmorepaka, 2020-03-02:
As long as your regions are in the same alignment as stock, you only need to flash the `bios` region. 

You can verify this by taking a backup of the `fd` region internally and inspecting it with [UEFITool](https://github.com/LongSoft/UEFITool/releases). Clicking on the Descriptor tab will show the description on the side and if the `BIOS region offset` is `500000h` You are good to go.

To flash it you have to get an image of it and pad it by 8M.

 You can either extract the untouched image from the update package by running `dd if=STOCK.FL1 bs=1 of=4MB.bin skip=464 count=4194304` or using the [1vyrain image](https://github.com/n4ru/patched-bioses/raw/84dc0b5bfaa95fccada26deb8eebd19943dd7d11/X230.rom) which is already trimmed and doesn't have whitelists. 

Whichever one you choose you have to pad it, `dd if=/dev/zero of=pad.bin bs=1M count=8` to create your padding, and `cat pad.bin 4MB.bin > 12M.rom`. 

You can flash the 12M.rom with the `--ifd -i bios` flag on flashrom.

NOTE: If you use 1vyrain image or extract from the latest lenovo firmware you will have to downgrade in addition to flash the EC, this can be done with `winflash.exe /sd /file [path to older .FL1]` or using the automated [IVprep](https://github.com/n4ru/IVprep) tool.

jaquer, 2021-01-05:
Hi @digmorepaka. I am in the same boat as OP: I installed skulls using 1vyrain, and while I love the concept, unfortunately the coreboot ACPI implementation is giving me some troubles. Can you please clarify this part of your instructions: 

> You can verify this by taking a backup of the fd region internally

How would I go about performing this step?

Also, when running `flashrom` is the [`iomem=relaxed`](https://github.com/merge/skulls/tree/master/x230#updating) kernel parameter necessary?

Thanks in advance for your help.

jaquer, 2021-01-05:
OK, I did some more research and was able to figure out how to do it. You definitely need to pass the `iomem=relaxed` kernel parameter before performing all these:

1. Run `flashrom -p internal -r ./fd.rom --ifd -i fd`
2. Follow the rest of the steps [here](https://github.com/merge/skulls/issues/137#issuecomment-593675096).
3. The final step is to run `flashrom -p internal:laptop=force_I_want_a_brick -w 12M.rom --ifd -i bios -N`.
4. Cross your fingers.

In my case, I ran into a couple of hiccups (RTC and HD detection issues), but those are OS problems. As far as I can tell, the switch from skulls to the 1vyrain BIOS worked perfectly.

mcanevet, 2021-04-25:
@jaquer I'd also like to rollback from skulls to 1vyrain. I'm wondering what's the purpose of running `flashrom -p internal -r ./fd.rom --ifd -i fd` as you don't use the resulting file (`fd.rom`) afterwards unless I missed something...

jaquer, 2021-04-25:
@mcanevet: You use the `fd.rom` file to verify that the existing ROM regions are in alignment, as outlined [in these steps](https://github.com/merge/skulls/issues/137#issuecomment-593675096).

I honestly don't know enough about flashing, etc, but it sounds like it's a sanity check before actually performing the flash. To expand on my instructions, you would:

1. Run `flashrom -p internal -r ./fd.rom --ifd -i fd` to extract the `fd` region of your currently installed BIOS.
2. Load the resulting `fd.rom` file into [UEFITool](https://github.com/LongSoft/UEFITool/releases).
3. Click on the Descriptor tab and confirm that `BIOS region offset` is `500000h`. (Unfortunately, I don't know what you would do if it doesn't match).
4. Extract the untouched image from the stock BIOS update package by running `dd if=STOCK.FL1 bs=1 of=4MB.bin skip=464 count=4194304` or using the [1vyrain image](https://github.com/n4ru/patched-bioses/raw/84dc0b5bfaa95fccada26deb8eebd19943dd7d11/X230.rom). No need to extract, since it's already trimmed, you just have to rename it `4MB.bin`.
5. Pad either image to 12MB by running `dd if=/dev/zero of=pad.bin bs=1M count=8` to create your padding and `cat pad.bin 4MB.bin > 12M.rom` to concatenate the base image and the padding.
6. Finally run `flashrom -p internal:laptop=force_I_want_a_brick -w 12M.rom --ifd -i bios -N` to flash.
7. Cross your fingers.

Hope that helps.

mcanevet, 2021-04-28:
I worked, thanks a lot.