merge/skulls #138: Clarification of RPi3 hardware example

k3tan172, 2020-02-09:
In the diagram of the README, there are 2 GND pins and a TX and an RX pins. Is the GND/TX/RX that are next to each other required to be used? The hardware requirements say only 6 jumper wires are required. Not using the GND/TX/RX that are next to each other is the only way this makes sense, unless I'm missing something.

Please advise.

jcholsap, 2020-02-09:
Personally, I find it easiest to use pins 17, 19, 21, 23, 25 and 24. I have a keyboard and monitor attached to my RPI and, so, don't need TX and RX.
[image]

k3tan172, 2020-02-09:
This is excellent. Thank you for clarification.

closed 2020-02-09 by k3tan172