merge/skulls #140: Included memtest86+ produces false positives

JenSte, 2020-02-24:
After installing skulls 0.1.8 on my x230 I ran the included memtest86+ and it consistently either hangs after a few seconds or detects a RAM error at 3071MB. The system contains two different RAM modules (the original 4gig FRU 03X6561 and a 8gig Kingston KVR16S11/8), the hangs/errors also happen when only one of the two modules is installed or when they are put into different slots (but the system never had problems recognizing the modules, as some users on reddit reported).

To check if the RAM is really broken I then did the following:
- Ran memtest86 (without the plus) v4.3.7, it did several iterations without finding any errors.
- Recompiled skulls and included the memtest86+ v5.01 floppy image from memtest.org as described at [1], then flashed it onto the "top" chip. It also ran without finding any errors.
- Booted Linux, which then ran stable, which would probably not be the case if the RAM is bad.

This is probably nothing skulls can do a lot about, but some kind of miscommunication between coreboot, seabios and the coreboot version of memtest86+ that's been there for quite some time (see also [2] and [3]). Maybe it would be better to not include the broken memtest86+ as payload but point users to a working one in the readme file?

[1] https://mail.coreboot.org/pipermail/coreboot/2018-November/087713.html
[2] https://www.reddit.com/r/coreboot/comments/amonz5/135v_ram_compatibility_and_some_research/efo2hp1/
[3] https://mail.coreboot.org/pipermail/seabios/2018-November/012559.html

Barraguesh, 2020-03-17:
Same here. The original BIOS didn't give any errors in my already included 2x4Gb sticks. I ran memtest86 from the BIOS but it pauses at 54%, giving some errors at 3071Mb, but same happens with a new Samsung 8Gb stick. The thing is that it also happens with Memtest86 on Ubuntu live image, doesn't froze but shows same 3071Mb errors, maybe it's a bug in the current version of memtest? Gonna check with the live image and in another system but I don't feel anything under normal use.

merge, 2020-03-18:
I'm thinking we should remove memtest again from our image.

Th3Fanbus, 2020-05-29:
Upstream we're aware that memtest86+ as coreboot payload does not seem to read the memory map properly, so it tests reserved memory regions. Nobody stepped in to debug or fix it though.

merge, 2020-06-30:
we removed memtest from the build again

closed 2020-06-30 by merge