merge/skulls #142: Reading old flash cihp contents - transaction error between offset 0x00003000 and 0x0000303f

kahrkunne, 2020-03-02:
I accidentally made my bootsplash image the wrong resolution when compiling skulls, so it doesn't display. No problem, I thought - I'll just replace it with a good one using the "updating" instructions. However, when I try to do that, this happens:

```[kahr@Manabi x230]$ du -h ~/top.rom 
4.0M	/home/kahr/top.rom
[kahr@Manabi x230]$ sudo cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=12295e63-8c3f-47ee-9458-a9ea9ed83d03 rw quiet splash loglevel=3 rd.udev.log-priority=3 vt.global_cursor_default=0 iomem=relaxed
[kahr@Manabi x230]$ ./x230_skulls.sh -U
You are using the latest version of Skulls for the X230
[kahr@Manabi x230]$ sudo ./x230_skulls.sh -i ~/top.rom 
input: top.rom
output: output/top_prepared_12mb.rom
WARNING: Make sure not to power off your computer or interrupt this process in any way!
         Interrupting this process may result in irreparable damage to your computer!
util/functions.sh: line 37: [: 98
26: integer expression expected
Flash the BIOS now? y/N: y
flashrom v1.2 on Linux 5.4.6-arch3-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using region: "bios".
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xbff51000.
Found chipset "Intel QM77".
Enabling flash write... SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG2: Management Engine region (0x00003000-0x004fffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing due to multiple flash chips detected.
OK.
Found Programmer flash chip "Opaque flash chip" (12288 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading old flash chip contents... Transaction error between offset 0x00003000 and 0x0000303f (= 0x00003000 + 63)!
FAILED.
```
In case it's relevant, I used 1vyrain to flash Skulls.

Thrilleratplay, 2020-03-02:
@kahrkunne It looks like your BIOS is not unlocked.  This is an issue for [1vyrain](https://github.com/n4ru/1vyrain) but I suspect you will need to flash externally to fix this.

digmorepaka, 2020-03-02:
This happens because the `me` region is not read-able(as noted in the flashrom log `FREG2: Management Engine region (0x00003000-0x004fffff) is locked.`). The error clearly states the error occurs at `0x00003000` which just so happens to be the start of `me` region. It's nowhere near the `bios` region. `bios` only starts at `0x00500000`. This can be solved by using the `-N` option for flashrom which skips integrity checks.

kahrkunne, 2020-03-03:
Ah, I see, that's the "--noverify-all switch" it was talking about. I did look this up when I originally got that message, but I looked up the flashrom manpage on a difference machine which had an older version which did not include this switch. I added that switch to the x230_skulls.sh script and can confirm that it now works properly!

merge, 2020-03-27:
thanks for reporting! should be fixed.

commit 106d9ff1 (2020-03-27): x230: add flashrom options similar to what Heads does

Fixes #142
closed 2020-03-27 by merge