merge/skulls #143: weird behaviour after flash - no hdd boot, keeps booting to memtest

MrSlimbrowser, 2020-03-03:
Hi,
I tried to install it as described and didn't see any errors but it doesn't seem to work.
The first poweron after flash made the keyboard toplight flash 3 times, then it restarted itself. Now when I try to poweron it shows the SeaBIOS (version rel-*) screen with the machine UUID on it, then starts memtest. If I hit ESC I get a bootmenu with following options:
1. AHCI/0: Samsung SSD* - selecting this one starts memtest
2. AHCI/2: MXSSD* - Selecting this shows "Booting from HDD" and a blinking cursor, nothing else
3. Payload [memtest] - starts memtest
4. Payload [nvramcui] - brings up some menu, probably the one intended
5. Payload [coreinfo] - brings up some menu, probably the one intended

I updated the BIOS using latest Lenovo original file first. Then I flashed using a RPi 3B, bottom first and then top.
I uploaded the commands outputs from installing [here (pastebin.com)](https://pastebin.com/xgBcZae5)
The part where it says "chip not detected" was my fault. At first, I accidentaly had Pin3 connected to MISO instead of Pin2 when trying to flash top.

Do you have any idea what may have gone wrong? Should I flash something a second time or do I have to revert using the backup?

mszumilak, 2020-03-03:
coreboot/heads/skulls cannot boot from GPT. Do you have msdos or gpt label on your hdd?

MrSlimbrowser, 2020-03-03:
oh my.. so much text for such an easy answer...
Yes, both hdds use gpt partition style, tried booting from usb which works fine. I'll reinstall my system and report back if everything is working (or not working).
Thank you

paulmenzel, 2020-03-03:
Just to clarify, it is a “design limitation” of the payload *SeaBIOS*, which emulates a standard BIOS.

@MrSlimbrowser, you could dry a different payload like GRUB or TianoCore.

MrSlimbrowser, 2020-03-03:
Oh nice, so there is something I can do about it. I'll try that, thank you for clarifying!

paulmenzel, 2020-03-03:
Actually, I was told, the BIOS (SeaBIOS) doesn’t care about the partition scheme. It just copies the MBR to RAM and jumps to it. So, MrSlimbrowser you should check your MBR. Maybe booting a live system and running `grub-install` will do the trick.

How did you boot your system with the vendor fimware?

Using GRUB as payload would still be able to boot your current system. You can also add it as a secondary payload first, and load it from SeaBIOS.

mszumilak, 2020-03-03:
> Actually, I was told, the BIOS (SeaBIOS) doesn’t care about the partition scheme. It just copies the MBR to RAM and jumps to it. So, MrSlimbrowser you should check your MBR. Maybe booting a live [...]

yes, but you have to have MBR which is not created with UEFI/GPT setup

paulmenzel, 2020-03-03:
> yes, but you have to have MBR which is not created with UEFI/GPT setup

What part does your *yes* refer too?

To clarify, using GRUB as payload (or loading it from CBFS (`img/grub.elf` with SeaBIOS), you do not need an MBR, as GRUB is already the boot loader and brings drivers to read the required files, for example, from an HDD.

paulmenzel, 2020-03-03:
https://wiki.archlinux.org/index.php/GRUB#GUID_Partition_Table_(GPT)_specific_instructions might help you with setting up the MBR.

MrSlimbrowser, 2020-03-04:
Okay, so I read and tried quite a lot and got grub2 running as payload just fine. No luck with tianocore though, couldn't get it to compile without errors. I noticed though that coreboot doesn't offer me anything I really need, it just makes most things I need more complicated. I learned a lot and it was a nice journey, but now I'm back to stock.
Thanks both of you for helping and leading me to the right direction :)

closed 2020-03-04 by MrSlimbrowser