merge/skulls #144: Only one RAM stick recognized with DDR3L

jm2, 2020-03-10:
2x8gb DDR3L; only 1x8gb shows up on both an x230 and x230t. I believe what I'm seeing is similar to this issue: https://www.reddit.com/r/coreboot/comments/amonz5/135v_ram_compatibility_and_some_research/

So it possible this may be worked around with some simple compilation tweaks.

digmorepaka, 2020-03-22:
Did you look at cbmem and seeing if there are any errors regarding memory? Do you see both spds on the smbus with i2c-tools? Did you try the sticks individually? 

Regarding the reddit post, before coreboot 4.9 native ram init was very unreliable, resuming your machine from suspend was a russian roulette of sorts. That reddit post is pretty old and most of it doesn't apply anymore. 
Coreboot runs 1.35V ram in it's 1.5V compatibility mode, as allowed by jedec.

jm2, 2020-04-06:
I did reseat the RAM which had no effect, however considering I have two different machines with two different sets of RAM, I doubt that this is something other than a bug in the RAM detection logic.

dmidecode:
```
$ sudo dmidecode -t memory
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.
 
Handle 0x000A, DMI type 17, 40 bytes
Memory Device
    Array Handle: 0x0000
    Error Information Handle: Not Provided
    Total Width: 64 bits
    Data Width: 64 bits
    Size: 8192 MB
    Form Factor: SODIMM
    Set: None
    Locator: Channel-0-DIMM-0
    Bank Locator: BANK 0
    Type: DDR3
    Type Detail: Synchronous
    Speed: 800 MT/s
    Manufacturer: Unknown (9b0d)
    Serial Number: ffffffff
    Asset Tag: Not Specified
    Part Number: None
    Rank: 2
    Configured Memory Speed: 800 MT/s
    Minimum Voltage: Unknown
    Maximum Voltage: Unknown
    Configured Voltage: Unknown
```

i2c-tools:
```
$ decode-dimms
# decode-dimms version $Revision$
 
Memory Serial Presence Detect Decoder
By Philip Edelbrock, Christian Zuckschwerdt, Burkart Lingner,
Jean Delvare, Trent Piepho and others
 
 
Decoding EEPROM: /sys/bus/i2c/drivers/eeprom/0-0050
Guessing DIMM is in                              bank 1
 
---=== SPD EEPROM Information ===---
EEPROM CRC of bytes 0-116                        OK (0xB729)
# of bytes written to SDRAM EEPROM               176
Total number of bytes in EEPROM                  256
Fundamental Memory type                          DDR3 SDRAM
Module Type                                      SO-DIMM
 
---=== Memory Characteristics ===---
Maximum module speed                             1600 MHz (PC3-12800)
Size                                             8192 MB
Banks x Rows x Columns x Bits                    8 x 16 x 10 x 64
Ranks                                            2
SDRAM Device Width                               8 bits
Bus Width Extension                              0 bits
tCL-tRCD-tRP-tRAS                                11-11-11-28
Supported CAS Latencies (tCL)                    11T, 10T, 9T, 8T, 7T, 6T, 5T
 
---=== Timings at Standard Speeds ===---
tCL-tRCD-tRP-tRAS as DDR3-1600                   11-11-11-28
tCL-tRCD-tRP-tRAS as DDR3-1333                   10-10-10-24
tCL-tRCD-tRP-tRAS as DDR3-1066                   8-8-8-19
tCL-tRCD-tRP-tRAS as DDR3-800                    6-6-6-14
 
[... 89 lines cut ...]
Auto Self-Refresh?                               Yes
On-Die Thermal Sensor readout?                   No
Partial Array Self-Refresh?                      No
Module Thermal Sensor                            No
SDRAM Device Type                                Standard Monolithic
 
---=== Physical Characteristics ===---
Module Height                                    30 mm
Module Thickness                                 2 mm front, 2 mm back
Module Width                                     67.6 mm
Module Reference Card                            F revision 0
Rank 1 Mapping                                   Standard
 
---=== Manufacturer Data ===---
Module Manufacturer                              Unknown
DRAM Manufacturer                                Micron Technology
Manufacturing Location Code                      0x01
Manufacturing Date                               2012-W28
Part Number                                      Undefined
 
 
Number of SDRAM DIMMs detected and decoded: 2
```

cbmem logs will be added as soon as I compile the binary to read them.

jm2, 2020-04-06:
cbmem console output:
```
$ sudo cbmem -c

*** Pre-CBMEM romstage console overflowed, log truncated! ***
g Ivybridge RAM training (1).
100MHz reference clock support: yes
Trying CAS 11, tCK 320.
Found compatible clock, CAS pair.
Selected DRAM frequency: 800 MHz
Selected CAS latency   : 11T
PLL busy... done in 50 us
MCU frequency is set at : 800 MHz
Done dimm mapping
Update PCI-E configuration space:
PCI(0, 0, 0)[a0] = 0
PCI(0, 0, 0)[a4] = 2
PCI(0, 0, 0)[bc] = c2a00000
PCI(0, 0, 0)[a8] = 3b600000
PCI(0, 0, 0)[ac] = 2
PCI(0, 0, 0)[b8] = c0000000
PCI(0, 0, 0)[b0] = c0a00000
PCI(0, 0, 0)[b4] = c0800000
PCI(0, 0, 0)[7c] = 7f
PCI(0, 0, 0)[70] = fe000000
PCI(0, 0, 0)[74] = 1
PCI(0, 0, 0)[78] = fe000c00
Done memory map
Done io registers
t123: 1767, 6000, 7620
ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : Bring up
ME: Current Operation Mode  : Normal
ME: Error Code              : No Error
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : Waiting for DID BIOS message
ME: FWS2: 0x101f0126
ME:  Bist in progress: 0x0
ME:  ICC Status      : 0x3
ME:  Invoke MEBx     : 0x0
ME:  CPU replaced    : 0x0
ME:  MBP ready       : 0x1
ME:  MFS failure     : 0x0
ME:  Warm reset req  : 0x0
ME:  CPU repl valid  : 0x1
ME:  (Reserved)      : 0x0
ME:  FW update req   : 0x0
ME:  (Reserved)      : 0x0
ME:  Current state   : 0x1f
ME:  Current PM event: 0x0
ME:  Progress code   : 0x1
PASSED! Tell ME that DRAM is ready
ME: FWS2: 0x10500126
ME:  Bist in progress: 0x0
ME:  ICC Status      : 0x3
ME:  Invoke MEBx     : 0x0
ME:  CPU replaced    : 0x0
ME:  MBP ready       : 0x1
ME:  MFS failure     : 0x0
ME:  Warm reset req  : 0x0
ME:  CPU repl valid  : 0x1
ME:  (Reserved)      : 0x0
ME:  FW update req   : 0x0
[... 800 lines cut ...]

Press ESC for boot menu.

Turning on vga text mode console
SeaBIOS (version rel-1.13.0-0-gf21b5a4)
Machine UUID 60603d81-5212-11cb-a5da-889ab1ffde28
Searching bootorder for: HALT
drive 0x000f66e0: PCHS=16383/16/63 translation=lba LCHS=1024/255/63 s=488397168
Space available for UMB: d0000-ed800, f5fa0-f66e0
Returned 163840 bytes of ZoneHigh
e820 map has 9 items:
  0: 0000000000000000 - 000000000009fc00 = 1 RAM
  1: 000000000009fc00 - 00000000000a0000 = 2 RESERVED
  2: 00000000000f0000 - 0000000000100000 = 2 RESERVED
  3: 0000000000100000 - 00000000bff03000 = 1 RAM
  4: 00000000bff03000 - 00000000c2a00000 = 2 RESERVED
  5: 00000000f0000000 - 00000000f4000000 = 2 RESERVED
  6: 00000000fed40000 - 00000000fed45000 = 2 RESERVED
  7: 00000000fed90000 - 00000000fed92000 = 2 RESERVED
  8: 0000000100000000 - 000000023b600000 = 1 RAM
enter handle_19:
  NULL
Booting from Hard Disk...
Booting from 0000:7c00
```

Th3Fanbus, 2020-05-29:
The interesting part got cut off, because it doesn't fit in pre-RAM console buffer. Basically what happened is that raminit failed and one channel got disabled. You can get complete logs by using flashconsole:

- Enable flashconsole in Kconfig
- Flash `coreboot.rom` to the flash chips
- Power off
- Power on
- Read back the flashchips into `coreboot_with_log.rom`
- Use cbfstool to extract the log from `coreboot_with_log.rom`

gjikia, 2020-08-18:
@jm2 
Did you manage to solve this problem?
I am having the same issue: only one module from 2x8GB of G.Skill, DDR3L-1600 SO-DIMM PC3L-12800S is recognized by coreboot/skulls. Recompiling coreboot  with the option IGNORE_XMP_MAX_DIMMS mentioned on reddit did not help unfortunately.

jm2, 2020-08-18:
Unfortunately not yet. I don't use the ThinkPads frequently so the priority of building a custom coreboot image to debug this is pretty low for me. If someone has a prebuilt and tested logging-enabled coreboot image for the x230 or x230t and can be safely flashed onto an ivyra1n-unlocked thinkpad, I'm happy to test and report back.

freed00m, 2020-10-25:
One day, my DDR3 stick also stoped working and I've discovered it was just oxidization on the pins.
Few round with alcohol around the pins on the ram fixed it and 2nd RAM stick started to load again.

jm2, 2021-03-23:
I can confirm that a second flash on both machines (upgrading to the latest release) was able to resolve the issue on both systems where re-seating the RAM didn't work. It's hard to say if a newer release fixed the issue, or if the upgrade sorted out some flash anomaly from the initial conversion to skulls. It looks like this solution works for quite a few cases like this according to https://github.com/merge/skulls/issues/116 and others.

closed 2021-03-23 by jm2