merge/skulls #145: X230 Tablet (Type 3435) USB Issues

late-in-autumn, 2020-03-13:
There are actually two USB-related issues, so I figured that it might help by submitting them together.
If that is not the case, please let me know. Both of these issues are reproduced on my ThinkPad X230 Tablet (Type 3435) with the Skulls build 0.1.8 binary downloaded from the release page.

Issues 1: Skulls does not recognize attached USB sticks for booting if the system is cold started.

Steps to reproduce:
- Completely power off the system
- Attach a USB startup stick to any of the USB ports
- Press Esc when prompted to select the boot device
- The USB stick is not listed

Expected:
- The USB stick is recognized and can be used to boot the system

Workaround:
- When the USB stick is not shown, press T to enter TPM configuration
- Press Esc to exit TPM configuration menu (and the system reboots)
- Then press Esc again to select the boot device
- Now the USB stick is recognized and can be used to boot the system

Additional info: I tried this with a Micro Center USB 3.0 stick that has a LED activity indicator, and when cold booting, the LED does not lit at all, therefore I suspect the USB ports are not powered at all, but I do not have the proper electrical equipment to confirm this.

Issue 2: USB 3.0 Ports (on the left side) seem to operate only at USB 2.0 speeds (confirmed under Windows 10, have not tested under Linux yet).

Steps to reproduce:
- Insert any USB 3.0 device into one of the USB 3.0 port (I have tried the same Micro Center USB stick, and a ThinkPad USB 3.0 dock with a Lexar USB 3.0 card reader attached)
- Go to Settings - Devices
- Windows prompts that "the device can operate faster if attached to a USB 3.0 port"
- I have also confirmed this by inserting a professional-grade SD card (Sony SF-G 32GB) into the Lexar reader and use CrystalDiskMark to test the I/O speeds, and it is indeed capped to about 20MB/s

Expected:
- USB 3.0 devices should operate at USB 3.0 speeds

I have checked both the old Coreboot wiki and the new docs.coreboot.org sites for information on X230 and X230 Tablet but cannot find information on these issues. In fact, https://www.coreboot.org/Board:lenovo/x230#Status actually states that USB is "working" for both 2.0 and 3.0 ports.

Thanks you.

late-in-autumn, 2020-03-13:
I have tested again, and it seems that all three USB ports on the machine are capped to USB 2.0: I cannot get 3.0 speed from either the two 3.0 ports on the left or the yellow "always-on" USB-eSATA combo port on the right. I have also tried removing the Slice Battery but it did not help.

I have also confirmed that the USB 3.0 port is recognized:
[image]
Sorry I do not have an English version available, but this basically says that an Intel USB xHCI controller is available on PCI 0,20,0 and is running on the Microsoft driver.

Sincerely

merge, 2020-07-01:
this should now work when using a skulls-x230t release ( https://github.com/merge/skulls/releases/tag/x230t-0.0.1 )