merge/skulls #147: X230 Ultrabase 3 issue

larghio, 2020-03-21:
Everything works when docked except undock button. undock light and my hardrive  that I installed in the dock in a ultrabay hdd caddy. dvd also does not work. confirmed works with lenovo bios.

digmorepaka, 2020-03-22:
Undock button is not needed with any modern OS on this machine. Every interface on it is hotplug capable by spec. In the devicetree.cb the dock sata is enabled, try doing a clean build and trying again. If you happen to have a 4338 pr 4337 desktop dock with eSata, try if it's eSata works.

larghio, 2020-03-27:
Hi . I agree about the dock. Just a observation. I got my ultrabay hdd to show up.
If I do a cold boot it works. if I dock with the power on or in S3 suspend. No hdd. nothing hdd related in the logs.

closed 2020-04-10 by larghio