merge/skulls #148: Didn’t Boot after flashing skulls-x230-0.1.9

Ego-sum-qui-sum, 2020-03-27:
After I flashed skulls onto my x230 laptop, it doesn't boot anymore. When I press the power-button, the led lights up for 1 second, but there is no further response.
I flashed the backup of the old Bios wich is Version 2.73 (G2ETB3WW) but the result is the same.

The backup I made is ok because I made 3 copies from the 4 and 8 mb chip and all have the same md5sum.

I used a ch341a Flasher with a voltage mod to 3.3V, because original it uses 5V wich is too much for the chips https://www.eevblog.com/forum/repair/ch341a-serial-memory-programmer-power-supply-fix/msg1323775

On flashing I controlled the voltage on pin 8 and it is exact 3.3V. I used USB3 port the current is enough. The AMS1117 on the flasher can handle 1A what is pretty enough. The chip has a power consumption at programming of typical 15mA

Lsusb is giving this result: Bus 005 Device 003: ID 1a86:5512 QinHeng Electronics CH341 in EPP/MEM/I2C mode, EPP/I2C adapter.

I disconnect power and the battery and the RTC battery from the laptop.

Another guy has described a similar problem at issues #97 what is unfortunately unsolved.

I hope somebody can help me.

Here the result from flashing:
Outputdata from flashing skull bottom chip *1)
Outputdata from flashing skull top chip *2)
Outputdata from flashing lenovo Bios 8mb *3)
Outputdata from flashing lenovo Bios 4mb *4)

*1)
sudo ./external_install_bottom.sh -m
Skulls for the X230

Please select the hardware you use:
1) Raspberry Pi
2) CH341A
3) Exit
Please select the hardware flasher: 2
Ok. Connect a CH341A programmer
trying to detect the chip...
Detected MX25L6406E/MX25L6408E.
make: Verzeichnis „/home/adonis/skulls-x230-0.1.9/util/ifdtool“ wird betreten
gcc -O2 -g -Wall -Wextra -Wmissing-prototypes -Werror -I../commonlib/include -c -o ifdtool.o ifdtool.c
gcc -o ifdtool ifdtool.o 
make: Verzeichnis „/home/adonis/skulls-x230-0.1.9/util/ifdtool“ wird verlassen
Intel ME will be cleaned.
The flash ROM will be unlocked.
Start reading 2 times. Please be patient...
flashrom v0.9.9-r1954 on Linux 4.15.0-91-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) on ch341a_spi.
Reading flash... done.
flashrom v0.9.9-r1954 on Linux 4.15.0-91-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) on ch341a_spi.
Reading flash... done.
connection ok
start unlocking ...
Full image detected
The ME/TXE region goes from 0x3000 to 0x500000
Found FPT header at 0x3010
Found 1 partition(s)
Found FTPR header: FTPR partition spans from 0x180000 to 0x24a000
ME/TXE firmware version 8.1.0.1265
Public key match: Intel ME, firmware versions 7.x.x.x, 8.x.x.x
The AltMeDisable bit is SET
Reading partitions list...
 FTPR (0x00180000 - 0x00024a000, 0x000ca000 total bytes): NOT removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xef)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x1cc428 - 0x1cc5e6       ): removed
 ROMP             (Huffman, fragmented data, ~ 2 KiB   ): NOT removed, essential
 BUP              (Huffman, fragmented data, ~ 56 KiB  ): NOT removed, essential
 KERNEL           (Huffman, fragmented data, ~ 135 KiB ): removed
 POLICY           (Huffman, fragmented data, ~ 91 KiB  ): removed
 HOSTCOMM         (LZMA   , 0x1cc5e6 - 0x1d3302       ): removed
 RSA              (LZMA   , 0x1d3302 - 0x1d8557       ): removed
 CLS              (LZMA   , 0x1d8557 - 0x1ddce8       ): removed
 TDT              (LZMA   , 0x1ddce8 - 0x1e43cd       ): removed
 FTCS             (Huffman, fragmented data, ~ 18 KiB  ): removed
 ClsPriv          (LZMA   , 0x1e43cd - 0x1e47ae       ): removed
 SESSMGR          (LZMA   , 0x1e47ae - 0x1f30b7       ): removed
The ME minimum size should be 1667072 bytes (0x197000 bytes)
The ME region can be reduced up to:
 00003000:00199fff me
Setting the AltMeDisable bit in PCHSTRP10 to disable Intel ME...
Removing ME/TXE R/W access to the other flash regions...
Checking the FTPR RSA signature... VALID
Done! Good luck!
File /tmp/tmp.SHJMVFfmp8/work.rom is 8388608 bytes
Writing new image to /tmp/tmp.SHJMVFfmp8/work.rom.new
ifdtool and me_cleaner ok
make: Verzeichnis „/home/adonis/skulls-x230-0.1.9/util/ifdtool“ wird betreten
rm -f ifdtool *.o *~ .dependencies
make: Verzeichnis „/home/adonis/skulls-x230-0.1.9/util/ifdtool“ wird verlassen
start writing...
flashrom v0.9.9-r1954 on Linux 4.15.0-91-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... 
Warning: Chip content is identical to the requested image.
Erase/write done.
DONE

*2)
sudo ./external_install_top.sh
1) ./x230_coreboot_seabios_4bd6927388_top.rom
2) ./x230_coreboot_seabios_free_4bd6927388_top.rom
3) ./x230_4mb.rom
4) Quit
Please select a file to flash or start with the -i option to use a different one: 1
Please select the hardware you use:
1) Raspberry Pi
2) CH341A
3) Quit
Please select the hardware flasher: 2
trying to detect the chip...
Detected MX25L3206E/MX25L3208E.
verifying SPI connection by reading 2 times. please wait.
flashrom v0.9.9-r1954 on Linux 4.15.0-91-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Macronix flash chip "MX25L3206E/MX25L3208E" (4096 kB, SPI) on ch341a_spi.
Reading flash... done.
flashrom v0.9.9-r1954 on Linux 4.15.0-91-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Macronix flash chip "MX25L3206E/MX25L3208E" (4096 kB, SPI) on ch341a_spi.
Reading flash... done.
connection ok. flashing x230_coreboot_seabios_4bd6927388_top.rom
flashrom v0.9.9-r1954 on Linux 4.15.0-91-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Macronix flash chip "MX25L3206E/MX25L3208E" (4096 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... 
Warning: Chip content is identical to the requested image.
Erase/write done.
DONE

*3)
sudo flashrom -p ch341a_spi -w x230_8mb_03.rom -c "MX25L6406E/MX25L6408E"
flashrom v0.9.9-r1954 on Linux 4.15.0-91-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.

*4)
sudo flashrom -p ch341a_spi -w x230_4mb_03.rom -c "MX25L3206E/MX25L3208E"
flashrom v0.9.9-r1954 on Linux 4.15.0-91-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Macronix flash chip "MX25L3206E/MX25L3208E" (4096 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.

fhvyhjriur, 2020-12-08:
You are using really outdated flashrom version (0.9.9 is from 2016). Please use at least flashrom 1.1.
If you still have the issue, you can try now heads-maximized. This skulls project here rely on preinstalled parts of the spi-chip. Heads just provide the ready, tested and working images for the top and bottom chip and dont rely on any software that was preinstalled on the two spi chips because it simply overwrite all of those.
https://github.com/osresearch/heads/pull/703

PS: You wrote "The backup I made is ok because I made 3 copies from the 4 and 8 mb chip and all have the same md5sum.". Please keep in mind, that md5 and sha are just for decoration, not for file-comparisation or anything else that have any real world functionality: https://sha2017.org/blog/sha-1-i-m-made-for-decoration

merge, 2021-03-27:
the issue is 1 year old and it might be worth flashing the latest skulls release with a recent flashrom.