merge/skulls #149: Native Raminit

jellolag, 2020-03-28:
Are you using: 
CONFIG_USE_NATIVE_RAMINIT=y
in the configs for building the x230 images,
or is it on by default?

Can we also get CONFIG_NATIVE_RAMINIT_IGNORE_MAX_MEM_FUSES=y in the configs if not already in use.

digmorepaka, 2020-04-02:
Native raminit is default. What is the point of the last two options on the X230? It doesn't have any frequency limiting fuses set, unlike previous gen ThinkPads, and there is no point in ignoring XMP. If you have dimms with an XMP profile, you likely want the tighter timings and higher speed. It has also been proven in the past that X230 is well capable of handling 2400mhz DDR3 provided you win silicon lottery on your memory controller.

jellolag, 2020-04-04:
Can you reference where it is set by default in coreboot source or documentation?
Memory fuses, how might I confirm your statement using my own machine?
How does one verify and test pushing the memory frequency to 2400mhz?

I have a single crucial 16 GB DDR3 module that is not recognized by the laptop even though it is within spec and using this coreboot bios.
 My ram is running at 1600mhz even though it is higher spec.

digmorepaka, 2020-04-04:
X230 cannot address 16GB dimms, you need broadwell ulv for that. If there were limiting fuses this wouldn't be possible on stock firmware: https://valid.x86.fr/kqpeqq
Just flash a higher than 1866 XMP profile and retrain.
2400mhz on an X230: https://valid.x86.fr/sy2lkm

Native raminit is default: https://review.coreboot.org/plugins/gitiles/coreboot/+/refs/heads/master/src/mainboard/lenovo/x230/Kconfig#7

Th3Fanbus, 2020-05-29:
Those raminit options usually result in overclocking. That is, they can make the hardware run faster than what it is supposed to be run at. It is *by no means* something that should be enabled by default.

16 GiB DDR3 modules are *not* within specifications because Sandy and Ivy bridge CPUs can only support at most 8 GiB per DIMM. There simply aren't enough address lines to make use of a 16 GiB DIMM with Sandy or Ivy Bridge CPUs. I think even Haswell and Broadwell still lack hardware support for 16 GiB DDR3 modules. If you want more memory, either use DDR4 or find a machine with 4 DIMM slots.

digmorepaka, 2020-05-29:
@Th3Fanbus Broadwell does support these high capacity dimms as mentioned in the compatibility sheet(https://www.intelligentmemory.com/fileadmin/download/compatibilitylist.pdf) and as proven by real running machines (http://valid.x86.fr/4dcrwa).

Th3Fanbus, 2020-05-29:
@digmorepaka hm, interesting. Does it support two of these at the same time, though? Intel says [the i5-5200U can do at most 16 GiB](https://ark.intel.com/content/www/us/en/ark/products/85212/intel-core-i5-5200u-processor-3m-cache-up-to-2-70-ghz.html).

digmorepaka, 2020-05-30:
While i have not been able to find any validations of Broadwell-U series running this memory in dualchannel, i wouldn't call it impossible just purely on the ark rating. Intel often doesn't go back to revalidate new memory modules on old platforms. Here is an example: 940XM running 32GB(4x8GB dimms) http://valid.x86.fr/h6ckf2 while on ark they list the limit of 8GB(https://ark.intel.com/content/www/us/en/ark/products/43127/intel-core-i7-940xm-processor-extreme-edition-8m-cache-2-13-ghz.html)

fhvyhjriur, 2020-12-08:
I think this issue can be closed, because there is no issue nor any benefit of changing those two settings for the raminit.

merge, 2021-03-27:
thanks for the clarifications here

closed 2021-03-27 by merge