merge/skulls #153: Wiring for Clip

Wheat43232, 2020-04-29:
**Question**: In what order do I place the wires from the RPI to the Clip?

______________________________________

I noticed that @jcholsap had a fairly concise explanation on which pins as well as which type of pins to use on the RPI [here](https://github.com/merge/skulls/issues/138#issuecomment-583830847). However, in what order are users supposed to add the wires from the RPI to the clip?

jcholsap, 2020-04-29:
That's really more of a personal philosophy as to what the most likely human error might be. Should I attach ground first? But then what if I accidentally contact the exposed part of the GND clip with an accidentally hot 5VDC lead? Etc... 

1. make sure there is no power to the device: laptop battery, RTC battery, A/C charger.
2. make sure your programmer is off and has no power source.
3. attach leads to the EEPROM first, then to your programmer next.
4. Double check your pin/lead match-ups and connections. Now check again.
5. Then, power on your programmer.
* My personal experience is that these EEPROMs are extremely hardy.
** Some folks insist on snagging power from WOL. At your own risk.

Wheat43232, 2020-04-29:
> That's really more of a personal philosophy as to what the most likely human error might be. Should I attach ground first? But then what if I accidentally contact the exposed part of the GND clip [...]

I think I may have phrased the question a bit incorrectly.
What I meant was that on the clip there are 8 pins:
[for example this clip]

[SOIC8 Clip](https://www.pomonaelectronics.com/sites/default/files/styles/product_image__500px_x_450px_/public/5250_1.jpg?itok=fO5kse0Q)

Does for instance pin 17 go to pin 1 on the clip, or in some other order?

Wheat43232, 2020-04-29:
> > That's really more of a personal philosophy as to what the most likely human error might be. Should I attach ground first? But then what if I accidentally contact the exposed part of the GND clip with an accidentally hot 5VDC lead? Etc...
> > ```
> > 1. make sure there is no power to the device: laptop battery, RTC battery, A/C charger.
> > 
> > 2. make sure your programmer is off and has no power source.
> > 
> > 3. attach leads to the EEPROM first, then to your programmer next.
> > 
> > 4. Double check your pin/lead match-ups and connections. Now check again.
> > 
> > 5. Then, power on your programmer.
> > 
> > 
> > * My personal experience is that these EEPROMs are extremely hardy.
> >   ** Some folks insist on snagging power from WOL. At your own risk.
> > ```
> 
> I think I may have phrased the question a bit incorrectly.
> What I meant was that on the clip there are 8 pins:
> [[for example this clip]](https://coinsh.red/u/clip.jpg)
> 
> Does for instance pin 17 go to pin 1 on the clip, or in some other order?

    That's really more of a personal philosophy as to what the most likely human error might be. Should I attach ground first? But then what if I accidentally contact the exposed part of the GND clip with an accidentally hot 5VDC lead? Etc...

    1. make sure there is no power to the device: laptop battery, RTC battery, A/C charger.

    2. make sure your programmer is off and has no power source.

    3. attach leads to the EEPROM first, then to your programmer next.

    4. Double check your pin/lead match-ups and connections. Now check again.

    5. Then, power on your programmer.

    * My personal experience is that these EEPROMs are extremely hardy.
      ** Some folks insist on snagging power from WOL. At your own risk.

I think I may have phrased the question a bit incorrectly.
What I meant was that on the clip there are 8 pins:
[for example this clip]

Does for instance pin 17 go to pin 1 on the clip, or in some other order?

Wheat43232, 2020-04-30:
@jcholsap Just to elaborate more on what the question pertains too:
I was able to eventually find a concise tutorial for Coreboot ([link](https://vimeo.com/177951809#t=1411s)), recommending a setup like this:

```
 Pin 1 (clip) --> Pin 24 (RPI)
 Pin 2 -- > Pin 21
 Pin 4 (GND) --> Pin 25
 Pin 5 --> Pin 19
 Pin 6 --> Pin 23
```

Then to flash, what appears to be only the top chip on the mother board.
However, does this setup still work with the RPI 3 B+ or RPI 4. Is there another recommended router for Skulls? How to power the motherboard during flashing, e.c.t.

- thanks