merge/skulls #155: Dump bios for TianoCore from Skulls?

HoshiYamazaki, 2020-05-13:
Heyo,
So what I want is to dump bios for compilation of Coreboot with TianoCore (I need Intel Descriptor Firmware and other blobs).
What I want to ask, I can safely dump bios while using Coreboot + SeaBIOS installed using Skulls (or - simpy /I use skulls/) from LiveUSB Linux? I need to use Tianocore since all my tripleboot OS'es are installed in EFI mode (before) - Windows 10, Arch, Hackintosh Catalina.

Thanks for advise, though!

HoshiYamazaki, 2020-05-13:
I tried to compile TianoCore using similar config to Skulls, and after hour or two, I got Error 2 (UEFIPAYLOAD.fd error 2), so I probably will give up and wait for ready solution in Skulls/1vyrain.

I failed, but my ThinkPad will not!

freed00m, 2020-10-25:
What do you mean by "dump" ?

The IFD can be extracted with ifdtool after you read the whole chip. If you never read the flashchip before it's probably still locked and you will need to do that externally.

HoshiYamazaki, 2020-10-27:
I don't need solution for that time, I know now I have ability to dump data via internal flashing. Gonna close the issue.

closed 2020-10-27 by HoshiYamazaki

freed00m, 2020-10-27:
@HoshiYamazaki Not true at all, the Intel Management engine will not allow you to read the whole flash AS IS.

With stock Lenovo x230 bios untouched (not unlocked externally) you will read something but part of the image will be crap.