merge/skulls #157: bios setting keep reseting back to default

darnrain, 2020-05-24:
Any one else have this problem where the bios settings in coreboot keep resting back to default? Do I need to get another cmos battery for my x230?

paulmenzel, 2020-05-28:
Please document, how you solved it.

darnrain, 2020-05-28:
Well it's been 4 days now and setting are still there. The only thing I can think of is, I compiled skulls with gcc 9.3.0, I have had issues with gcc 10.  It is posible that it was an issue and because I compiled skulls I got all the new updates, the coreboot devs may have fixed the issue.

also I pulled out my bluetooth card out of my laptop because I got an intel 7260 AC-wifi. After I pulled out the blutooth from my laptop my trackpoint seemed like it was butter smooth. Odd how a bleutooth card would affect the trackpoint. But maybe the bluetooth card was messing with coreboot somehow.

Keep you updated.

darnrain, 2020-05-31:
5-30-20 Still have the same settings in my bios I am not sure what I did to fix the problem but i am thinking it's the blutooth card that was causing issues.

I am going to put this in the solved issues.

darnrain, 2020-07-04:
Well never did fix the issue it looks like. I just checked and the bios settings have reverted back to default. If any one has any info how to fix this please post.

alexgill, 2020-09-22:
Anecdotally, I certainly encountered this behavior when I tried booting with Windows7, but since that was only a test (with failed), booting into various Linux kernels/distros, I haven't seen this problem once.

closed 2020-10-21 by darnrain

darnrain, 2020-10-30:
Okay so I think I fixed it. Knowing that it's a windows 10 issue. I looked at the device manager and seen a device named pci data acquisition and signal processing controller. It was not enabled and there was no driver running it. After some investigation I found it controls the cpu power I think. It is the Intel Dynamic Platform and Thermal Framework. So I did a search for that and found a driver on the lenovo web site. I had to manual install the drivers, that is go into device manager and update. So far so good I hope this fixes my issue. And helps any one else that has this issue.

darnrain, 2020-10-30:
Well the bios is still resting back to defaults after I boot into windows. But I think I did fix the sleep issue in windows 10. Some times when I sleep and wake up the computer just reboots. And the sleep icon on the x230 is still on. So my computer thinks it's asleep and its not.

Some how I think it's a problem with Linux and windows regarding the time. I think I will tell windows not to change the time automatically. See if that fixes the issue with bios setting getting reset.

darnrain, 2020-11-08:
Well it's been 10 days now and bios has not reverted back to default. So it was Windows 10 changing the time messing up the bios settings. That's so odd I would have never have thought it was because windows was changing the time.