merge/skulls #158: Error upgrading skulls on x230

edward-riley, 2020-06-09:
I already have skulls installed.
I installed skulls using 1vyrain.

When booting up manjaro with iomem=relaxed and attempting to upgrade my bios using the tarball from the releases section i get this error:

Done on thinkpad x230

1) ./x230_coreboot_seabios_4bd6927388_top.rom
2) ./x230_coreboot_seabios_free_4bd6927388_top.rom
3) Quit
file not specified. Please select a file to flash. Please read the README for details about the differences: 1
input: x230_coreboot_seabios_4bd6927388_top.rom
output: output/x230_coreboot_seabios_4bd6927388_top_prepared_12mb.rom
WARNING: Make sure not to power off your computer or interrupt this process in any way!
         Interrupting this process may result in irreparable damage to your computer!
Flash the BIOS now? y/N: y
flashrom v1.2 on Linux 5.6.15-1-MANJARO (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using region: "bios".
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xbff51000.
Found chipset "Intel QM77".
Enabling flash write... SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG2: Management Engine region (0x00003000-0x004fffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
Enabling hardware sequencing due to multiple flash chips detected.
OK.
Found Programmer flash chip "Opaque flash chip" (12288 kB, Programmer-specific) mapped at physical address 0x0000000000000000.
Reading old flash chip contents... Transaction error between offset 0x00003000 and 0x0000303f (= 0x00003000 + 63)!
FAILED.

merge, 2020-06-10:
it looks like IFD is not unlocked in flash. maybe 1vyrain did not do that?

merge, 2020-06-10:
from what I read over there https://github.com/merge/skulls/issues/136#issuecomment-582478666 I think if you installed via 1vyrain, you indeed have to stick to 1vyrain to update.

edward-riley, 2020-06-28:
alright, but doesn't skulls only supports BIOS and 1vyrain only support UEFI? I have spent the last 2 weeks attempting to find a solution, but i dont know if there is one.

edward-riley, 2020-06-28:
Fixed! I added -U as a perimeter on my flashrom script.

aeri, 2020-06-30:
> 
> 
> Fixed! I added -U as a perimeter on my flashrom script.

I tried it, but flashrom not support the parameter -U, could you specify how you got it?
Thank you :)

digmorepaka, 2020-07-07:
@aeri if you are getting the same error as in the OP refer to https://github.com/merge/skulls/issues/142#issuecomment-593670864