merge/skulls #159: X230T Tablet Mode ACPI

aeri, 2020-06-15:
When the X230T's screen is rotated, an ACPI call is made to detect that the screen has been rotated and activate tablet mode. This call is not made with a coreboot. I am reading the api calls with `acpi_listen`

I am using the latest version 0.1.9

It is possible include this in future versions?

merge, 2020-06-30:
the X230T now is a seperate device upstream in coreboot. I haven't checked what exactly the differences are and whether that ACPI call is made in the x230t, but plans are to create dedicated x230t images in the future.

aeri, 2020-06-30:
Thank you, in this proyect [thinkpad-scripts](https://thinkpad-scripts.readthedocs.io/en/latest/guides/additional-keys.html) there are info about the special features of the tablet version like the bezel key.

merge, 2020-07-01:
can you test using https://github.com/merge/skulls/releases/tag/x230t-0.0.1 that should work better on the tablet?

aeri, 2020-07-04:
I am tried to update related to issue #158. I installed the BIOS with 1vyrain but now the method does not works because 1vyrain only supports UEFI but not coreboot, now i am at an impasse. I want to test the version as soon as @edward-riley can help me with its method. Thank you

aeri, 2020-10-17:
Sorry for late reply, finally I have managed to update the BIOS to the new version released specifically for the X230T, I have had to modify the script with `-p internal:boardmismatch=force` to correct the model from X230 to X230T. The update has gone well but I still can't use the buttons on the X230T's display and the tablet mode detector, these ACPI calls still don't work.

X230T specific buttons:
[image]

perhaps these calls could be extracted from the original ROM or be detected in some other way?

Thanks for your support :)