merge/skulls #163: Impossible to upgrade to Skulls 0.1.10

engenteur, 2020-07-20:
I'm using Skulls since it's early release and I never experienced any problems with updating. I always use Archlinux booted up with the parameter "iomem=relaxed". After today's update my x230 was not booting up anymore (ThinkLight was powering on as well as the fan but no image on the screen and blinking simultaneously power and mute buttons). I was lucky enough to boot it up with keeping ESC key pressed while turning the computer on. I downloaded the archive with 0.1.10 again and reflashed thinking that the issue comes from corrupted archive but the results are the same. Finally I succeded to again bring my laptop back to life with a small help of the ESC key... After downflashing to 0.1.9 no more problems. What could be the problem ? Anyone experienced the same issue ? I'm using the computer as my daily so I prefer not to brick it. I'm flashing the non-free image because from time to time I need to boot into Windows. Thanks for any feedback ! :)

digmorepaka, 2020-07-24:
What memory do you run?

engenteur, 2020-07-27:
I post the output of dmidecode : 
`# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.

Handle 0x000A, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x0000
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 8192 MB
	Form Factor: SODIMM
	Set: None
	Locator: Channel-0-DIMM-0
	Bank Locator: BANK 0
	Type: DDR3
	Type Detail: Synchronous
	Speed: 800 MT/s
	Manufacturer: Kingston
	Serial Number: 4b29bf39
	Asset Tag: Not Specified
	Part Number: KHX1866C11S3L/8G
	Rank: 2
	Configured Memory Speed: 800 MT/s
	Minimum Voltage: Unknown
	Maximum Voltage: Unknown
	Configured Voltage: Unknown

Handle 0x000B, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x0000
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 8192 MB
	Form Factor: SODIMM
	Set: None
	Locator: Channel-1-DIMM-0
	Bank Locator: BANK 0
	Type: DDR3
	Type Detail: Synchronous
	Speed: 800 MT/s
	Manufacturer: Kingston
	Serial Number: 4c29c239
	Asset Tag: Not Specified
	Part Number: KHX1866C11S3L/8G
	Rank: 2
	Configured Memory Speed: 800 MT/s
	Minimum Voltage: Unknown
	Maximum Voltage: Unknown
	Configured Voltage: Unknown
`
The commercial name was : "Kingston Technology HyperX Impact 8GB 1600MHz DDR3L CL9 SODIMM 1.35V". I've already done Skulls updates with this memory and everything was working correctly. I should put back the original Samsung memory for updating ? Thank you !

digmorepaka, 2020-07-27:
The part number corresponds to 1866C11 XMP 1600C9 SPD kit, what processor does your X230 have? You may want to disable XMP if the cpu does not handle it, it is also possible that the particular version that skulls is on has issues with the 1866 strap.

engenteur, 2020-08-03:
Sorry for late response. I have an i7-3520M. I'll try again the next week with another RAM. If it's not helping then maybe the next release of Skulls will flash normally :)