merge/skulls #164: Linuxboot, UEFI support

Linvydas, 2020-08-17:
Are project developers considering adding UEFI or linuxBoot (NERF) support?

HoshiYamazaki, 2020-09-11:
I think UEFI support can be added by just providing TianoCore Skulls builds, I think it's not really that hard but I bricked my first X230 doing that so I rather would use ready builds. Also I don't own Corebooted X230 now (I own X230 with standard bios).

nedson, 2020-10-09:
@HoshiYamazaki how did you brick your x230? Couldn't you just reflash with original bios firmware?

HoshiYamazaki, 2020-11-04:
> @HoshiYamazaki how did you brick your x230? Couldn't you just reflash with original bios firmware?

It just doesn't turn on, before I had ability to turn it on but I tried to use external flasher and probably fried the motherboard. Just bad luck. Anyway, I got another X230, so I hope I will be able to use it, but I expect somebody will prepare ready Coreboot+Tianocore builds since I want to use UEFI systems (Windows/Linux/macOS tripleboot).

varac, 2020-11-11:
+1 for adding a Tianocore image.
I'd like to update my x230 System Firmware using [fwupd](https://fwupd.org/), but I'm not able to do so because I would need to boot into UEFI mode:
```

❯ fwupdmgr get-devices
...
│
├─BCM20702A0:
...
│   
├─Samsung SSD 860 EVO 1TB:
...
│   
└─System Firmware:
      Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
      Current version:     G2ET82WW (2.02 )
      Vendor:              LENOVO (DMI:LENOVO)
      Update Error:        Firmware can not be updated in legacy mode, switch to UEFI mode
      GUID:                230c8b18-8d9b-53ec-838b-6cfc0383493a
      Device Flags:        • Internal device
                           • Requires AC power
                           • Needs a reboot after installation
```

Dongerrs, 2020-11-29:
+1 for Tianocore

kedare, 2021-04-06:
So if I understand right, a x230 flashed with skulls cannot boot to an UEFI OS ? (that would explain why I can't boot to my existing system after flashing it)

jcholsap, 2021-04-06:
I think the default payload is SeaBIOS (MBR) and you would need something like Tianocore to locate EFI on the disk. @Thrilleratplay has scripts to build Skulls. But I don't know much about his scripts. For a small laptop, is there really a benefit to using EFI?

qubicrm, 2021-04-10:
In some use cases UEFI could be usefull: for example for safeboot install. (though, probably heads would be a better option for those scenarios)

Thrilleratplay, 2021-04-10:
@jcholsap skulls has its own build scripts that have shared parts with [Thrilleratplay/coreboot-builder-scripts](https://github.com/Thrilleratplay/coreboot-builder-scripts) however I try to maintain a generalized pattern to flash internally across different devices.  They also use SeaBIOS by default but allow the user to change the configuration.  Although, as far as I know, Tianocore is only stable in MrChromeBox's upstream branch and is not suggested for other coreboot devices.

HoshiYamazaki, 2021-04-20:
> So if I understand right, a x230 flashed with skulls cannot boot to an UEFI OS ? (that would explain why I can't boot to my existing system after flashing it)

Yes, that's true.

> I think the default payload is SeaBIOS (MBR) and you would need something like Tianocore to locate EFI on the disk. @Thrilleratplay has scripts to build Skulls. But I don't know much about his scripts. For a small laptop, is there really a benefit to using EFI?

Sure we can benefit from it, especially for Hackintoshes (if you're macOS person), or you have now existing OS installations based on EFI schema on your disk (or disks, since X230 can support two at once, thanks to mSATA).

Anyway, we got "other" solution to the problem, probably `osboot` (https://osboot.org/) will solve the existing issues (and not only with TianoCore since it provides multiple prebuilt coreboot distributions).

tickelton, 2021-08-14:
I was also caught a little off guard by the lack of EFI support and had to fix my bootloader after flashing Skulls.
Yes, I know it says right there on top of the README that Skulls uses SeaBIOS and thereby implicitly that it doesn't support EFI.
But maybe it would still be a good idea to explicitly state it in the guides and suggest to switch to a bootloader that supports legacy BIOS before flashing Skulls.

yaroze, 2021-10-20:
+1 for Tianocore :D

HoshiYamazaki, 2021-12-28:
@merge can you please provide TianoCore builds? As I see the osboot maitainer is not working towards it anymore, and I think all we must do is just prepare correct makefile, build it, test it and thats all (?)

h0m35t4r, 2022-01-08:
+1 for TianoCore. Interested in booting from NVMe without a bootloader

PSzczepanski1996, 2022-08-25:
Hey, is there any update of this?

Baitinq, 2022-12-17:
If anyone is still interested in this @MrChromebox provides a prebuilt TianoCore x230 image (https://www.reddit.com/r/coreboot/comments/yo0cqb/comment/ivbvvek/?utm_source=share&utm_medium=web2x&context=3)

sunoc, 2024-07-24:
+1 for Tianocore image! 

I'll see what I can do with the MrChromeBox in the mean time!