merge/skulls #167: x230_skulls.sh -U does not work anymore

Gusher123, 2020-11-02:
```
~/skulls-x230-0.1.11$ ./x230_skulls.sh -U
You seem to use a development version. Please use release package skulls-x230 0.0.2 for flashing.

```
Probably because 
```
~/skulls-x230-0.1.11$ curl -s https://api.github.com/repos/merge/skulls/releases/latest | grep browser_download_url | cut -d'"' -f4 | cut -d'/' -f9 | head -n 1
skulls-x230t-0.0.2.tar.xz

```
is reporting the latest x230t version in stead of the latest x230 version.

Gusher123, 2020-11-02:
Changing to

`
        UPSTREAM_FILE=$(curl -s https://api.github.com/repos/merge/skulls/releases | grep browser_download_url | grep skulls-x230- | cut -d'"' -f4 | c$

        UPSTREAM_VERSION=$(curl -s https://api.github.com/repos/merge/skulls/releases | grep browser_download_url | grep skulls-x230- | cut -d'"' -f4 $

`

Fixes the problem

merge, 2020-11-02:
thanks a lot. merged and updated for x230t too.

closed 2020-11-02 by merge