merge/skulls #173: flashrom refuses to flash

tjburrows, 2020-11-19:
I recently tried flashing 0.1.11 onto my x230 with the internal flash script x230_skulls.sh, and I got a new warning from flashrom, preventing me from flashing, which I have not seen when I have previously done this same procedure for previous skulls releases.  I am using flashrom v1.0 on Opensuse.  Maybe my flashrom updated and changed behavior?  Do I need to provide extra flashrom arguments like `internal:laptop=force_I_want_a_brick` or `internal:laptop=this_is_not_a_laptop` ?

> sudo ./x230_skulls.sh -i build/x230_coreboot_seabios_8053595370_top.rom
input: x230_coreboot_seabios_8053595370_top.rom
output: output/x230_coreboot_seabios_8053595370_top_prepared_12mb.rom
WARNING: Make sure not to power off your computer or interrupt this process in any way!
         Interrupting this process may result in irreparable damage to your computer!
Flash the BIOS now? y/N: y
flashrom v1.0 on Linux 5.3.18-lp152.50-default (x86_64)
flashrom is free software, get the source code at https://flashrom.org
Using region: "bios".
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xbff55000.
========================================================================
WARNING! You seem to be running flashrom on an unsupported laptop.
Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.
If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================
Aborting.
Error: Programmer initialization failed.

fhvyhjriur, 2020-12-08:
Could you try using some more recent software? Flashrom 1.0 from 2018 is somewhat outdated. Since then many things have changed. If possible please try latest code (latest code, not release) from flashrom and report again. With latest code also the output of the flashrom command with --verbose attached would be interesting.
Do you have a Thinkpad x230 or a x230t ?

tjburrows, 2020-12-11:
It is a Thinkpad x230.  First I tried using the latest x230_skulls.sh from master, which did not change anything.  Then, I upgraded my flashrom version to 1.2, and then it worked like expected.  It is probably worth noting somewhere what the minimum version of flashrom that is needed.  I don't know about any versions other than 1.0 and 1.2, but the former doesn't work and the latter works.  Thanks for the advice.

closed 2020-12-11 by tjburrows