merge/skulls #174: x230 i5 3320M - Firefox Tab Crashing

nlwhd, 2020-11-28:
This may not be the right place to post such an issue.  I apologize if it's not. 

After successfully flashing Skulls externally on x230 i5 3320M, Firefox tabs seem to crash at random.  I had never had a tab crash prior to the Skulls flash.  

This occurs in Ubuntu OS, and also within Fedora and Debian QubesOS AppVMs.  

Also, unexpected system restarts occassionally occur for both the Ubuntu OS and QubesOS.

tlaurion, 2020-11-29:
RAM issues with coreboot ? dmidecode output in dom0 for RAM modules? Abu issues found searching for thèse + coreboot?

nlwhd, 2020-12-03:
> RAM issues with coreboot ? dmidecode output in dom0 for RAM modules? Abu issues found searching for thèse + coreboot?
> -- Sent from my Android device with K-9 Mail. Please excuse my brevity.

Without investigating from within dom0, I went ahead and swapped out the RAM from 16G to 8G.  Seems to have solved the problem, as I haven't seen the issue arise since the swap.  

I wonder if this has to do with the quality of the 16G RAM.

tlaurion, 2020-12-03:
ram module connection, chips having overheated, initial bad ram, ram modules being different brand and specs... There is a lot of possibilities there.

What I meant by swapping ram modules was to unseat the modules, and flip them around to verify connection, brand and specs to eliminate the first hypothesis being a bad connection. Then you can test one memory module alone, the other alone, and replace the bad memory module. Hynix, samsung, critical. I had the chance to test a lot of different modules where issues arised principally when there was mismatches. 

Kernel hanging, applications misbehaving leave traces, and those traces are the ones to investigate, while a bad ram module should just be changed when isolated and found faulty.

gulafaran, 2020-12-04:
worth knowing, i had tons of issues on my x230 when trying to overclock my ram to 2133. it all ended up in lenghty discussions with coreboot devs and then a large amount of raminit code went into coreboot git, https://review.coreboot.org/q/topic:%2522sandybridge%2522+(status:open+OR+status:merged),50 pretty much all the ones from september 23rd , after those i finally got stable ram speeds, and in turn he also found other bugs and issues with the raminit code. so if possible ty coreboot from git.

merge, 2021-03-22:
looks like this is resolved. please re-open if not. thank you!

closed 2021-03-22 by merge