merge/skulls #175: Unlock IFD/ME on '8M-bottom' chip when only '4M-top' flashed

fhvyhjriur, 2020-12-01:
I flashed once skulls following the skulls-howto. There it was and is still written, that flashing the bottom is optional. And because the external flasher was not able to detect for some reason the 8M-chip(worked fine for the 4M) i skipped the 8M-chip because it was written optional and nowhere any warnings that for example future heads releases wont work. 
I gave then my external flasher to other people and dont have it any more.

When i update skulls i always see:
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG2: Management Engine region (0x00003000-0x004fffff) is locked.

Now i would really like to use the additional space and shrink the ME. Also heads is requiring this for more modern releases: https://github.com/osresearch/heads/issues/307

Is it in the meantime possible to unlock the IFD/ME without me having to order some external flasher again? I have fully unlocked 4M chip and can install whatever i need into there.

merge, 2020-12-07:
well. read what the optional 8m chip flashing would do: unlock the ifd. that's optional for installing Skulls, but of course you only can flash externally then. That's intended.

Should we improve the documentation? how?

fhvyhjriur, 2020-12-08:
> Should we improve the documentation? how?

After spending some more time into learning the backgrounds, i made two pull requests. One for the x230 and one for the x230t installation documentation.

closed 2021-01-02 by fhvyhjriur