merge/skulls #178: Simplify flashing the 8M bottom chip without relying on the preinstalled closed source software

fhvyhjriur, 2020-12-08:
Recently flashing the 8M bottom chip rely on closed source software parts that are preinstalled on the 8M bottom chip. Those closed source parts are then partly modified and kept on the 8M chip. This modification of this closed source software rely on additional software. It would be much more simple when it wont rely on any additional software and can be just flashed as a ready file. Then the installation steps would be exactly the same for top and bottom chip. The only required software would then be flashrom and nothing else.
Now it can be told that its more kind of simple to install the heads-maximized version then skulls when 8M and 4M chip should be flashed and its recommended to flash both.
Skulls can then provide same reduced ME, universal GBE and blobtool based IFD like heads-maximized does now in their bottom flash file.

fhvyhjriur, 2021-01-19:
There is kind of a solution here: https://github.com/merge/skulls/issues/172#issuecomment-761903700

@merge You can see in this thread that having precompiled bottom-chip images could also solve strange issues like the battery-drain one reported in https://github.com/merge/skulls/issues/172
Are you planing in releasing precompiled bottom-chip images to simplify many the installation process and removing the chance of such strange issues like reported when everyone have to create its own bottom image with ifdtool and me_cleaner?

merge, 2021-03-22:
Don't you have to fiddle in your Ethernet MAC address manually in that case?

fhvyhjriur, 2021-04-03:
Simply flashing two prebuild images would help to not have to care about any preinstalled things on SPI chips. The backup is then optional. I have met until now about 4 Thinkpads that had BIOS/UEFI issues or completely broken BIOS/UEFI. Overwriting the two chips with heads-maximized fixed all the software issues.
Heads-maximized have a generic MAC address all heads-maximized devices have. This is great for privacy. When people like to have their sticker-MAC address or any other MAC address they could change it anytime they want when the IFD is unlocked after the initial skulls-flash.