merge/skulls #182: Incorrect output using ./x230_skulls.sh -U for X230

boraboragoldfish, 2021-02-17:
I am about to install what I thought is the latest stable release of Skulls x230-0.1.11 for the X230. However, when I checked with the line code `./x230_skulls.sh -U`, it says the following:

**You seem to use a development version. Please use release package skulls-x230 0.0.2 for flashing.**

Is skulls-x230 0.0.2 really the latest stable build? I am a bit confused as it was released all the way back on Jan 18, 2018. I do see the x230t-0.0.2 build, but that is for specifically the X230t. Is this a message error or are we recommended to use the older X230 build? Thanks.

merge, 2021-02-19:
I think this should be fixed with todays' release

boraboragoldfish, 2021-02-20:
Awesome. Thanks so much!

closed 2021-02-20 by boraboragoldfish