merge/skulls #184: Reverting back to stock BIOS

accomplishednose, 2021-02-21:
Hello, I bought an x230 with skulls flashed bios from a friend. I have both backups of the lower and upper chip. How am I about to reflash the chips with the stock BIOS since I do not need skulls as I am using windows for work and the system is not behaving 'well' with windows installed? 

Is hooking up a pomona clip + raspberry and just flashrom-ing both chips with their original .rom/.bak files sufficient?

merge, 2021-03-28:
yes that should always work and is probably how I'd do it. in some cases, flashing internally can work too as https://github.com/merge/skulls/issues/137 suggests.

NicePics, 2022-11-10:
What if you have _no backups_ on this second hand machine and want to revert for updating the EC and ME?