merge/skulls #185: Big clock drift occurs on each reboot (x230, 0.1.12)

nix42, 2021-03-01:
Hello,

I thought this may be an OS issue but have been having clock drifts each time I boot up my PC such as the year being 2071 or the date being 26/164/2090, weird stuff like that. I selected the coretemp info entry and it just kept displaying the date constantly and nothing else, booted into qubesOS and the date is also wrong in there, try setting it in the OS but upon reboot it is not fixed.

system:
- 15 3320m
- 16GB ddr3 ram
-  500GB SATA SSD

jcholsap, 2021-03-01:
To start, I'd check the hardware RTC:
1. Remove the laptop battery 
2. Disconnect all cables.
3. At boot splash screen, press F1.
4. Set/verify time/date.
5. Save settings.
6. Power off.
7. Disconnect all power from the laptop for 10 seconds.
8. Reconnect power.
9. Reboot. Repeats Steps 3 & 4.
10. It's also possible the RTC battery is failing. To test for that, repeat Step 7 for eight hours.

If there's no indication of a RTC issue, then I'd suspect a NTP issue in the OS setup. The simplest way of testing that would be to:
1. disable the service
2. manually set the date/time
3. reset/verify date/time in RTC (Steps 1-8 from above)
4. boot into os and verify date/time
Thoughts: If you disable the NTP service and correctly set/verify the RTC date/time, then the OS will use the RTC date/time upon boot.

nix42, 2021-03-04:
steps 1-9 fixed the issue for me, thank you

closed 2022-07-20 by merge