merge/skulls #191: Skulls not updating with internal flash

K0ntrast, 2021-04-16:
Hello,

I am running Debian 10 stable on an X230 (23241J7). It currently runs Skulls v0.1.11 (non-free Intel VGA blob) which was initially installed by flashing externally. The battery whitelist patch was applied with the OEM BIOS present (v2.75, EC v1.14) before flashing both top and bottom BIOS chips with Skulls. It seems to run just fine.

Anyway, I want to update Skulls to the latest release (v1.0.0). I am running the "skulls.sh" script provided in the release and this is what the output is:

$ sudo ./skulls.sh -b x230 -i x230_coreboot_seabios_4de7610b32_top.rom
input: x230_coreboot_seabios_4de7610b32_top.rom
output: output/x230_coreboot_seabios_4de7610b32_top_prepared_12mb.rom
WARNING: Make sure not to power off your computer or interrupt this process in any way!
         Interrupting this process may result in irreparable damage to your computer!
WARNING: To prevent shutdowns, we recommend to only run this script when
         your laptop is plugged in to the power supply AND
         the battery is present and sufficiently charged (over 25%).
Continue anyways? (please do NOT!) y/N: y
Flash the BIOS now? y/N: y

flashrom  on Linux 4.19.0-16-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using region: "bios".
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0xbff4f000.
========================================================================
WARNING! You seem to be running flashrom on an unsupported laptop.
Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.

If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================
Aborting.
Error: Programmer initialization failed.

What is going wrong here? Is it to do with the flashrom version I have (it was sourced from the distro's main apt source)? I have ensured the "iomem=relaxed" option has been set as a kernel boot parameter.

jcholsap, 2021-04-16:
Looks like flashrom is throwing an error. You might want to try to use flashrom directly to get more information. I'd start with just trying to read the BIOS. The verbose flag can be helpful too.

K0ntrast, 2021-04-19:
> Looks like flashrom is throwing an error. You might want to try to use flashrom directly to get more information. I'd start with just trying to read the BIOS. The verbose flag can be helpful too.

Thanks for your reply. I had a sneaking suspicion t it would probably be something to do with the flashrom program.

Here is the output from the program just examining the BIOS.

$ sudo flashrom -p internal -V

> flashrom  on Linux 4.19.0-16-amd64 (x86_64)
> flashrom is free software, get the source code at https://flashrom.org
> 
> flashrom was built with libpci 3.5.2, GCC 8.2.0, little endian
> Command line (3 args): flashrom -p internal -V
> Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
> Initializing internal programmer
> Found candidate at: 00000500-00000510
> Found coreboot table at 0x00000500.
> Found candidate at: 00000000-00000ae0
> Found coreboot table at 0x00000000.
> coreboot table found at 0xbff4f000.
> coreboot header(24) checksum: 4c15 table(2784) checksum: 173e entries: 31
> Vendor ID: LENOVO, part ID: ThinkPad X230
> Using Internal DMI decoder.
> DMI string chassis-type: "Laptop"
> Laptop detected via DMI.
> DMI string system-manufacturer: "LENOVO"
> DMI string system-product-name: "23241J7"
> DMI string system-version: "ThinkPad X230"
> DMI string baseboard-manufacturer: "LENOVO"
> DMI string baseboard-product-name: "23241J7"
> DMI string baseboard-version: "ThinkPad X230"
> W836xx enter config mode worked or we were already in config mode. W836xx leave config mode had no effect.
> Active config mode, unknown reg 0x20 ID: 00.
> ========================================================================
> WARNING! You seem to be running flashrom on an unsupported laptop.
> Laptops, notebooks and netbooks are difficult to support and we
> recommend to use the vendor flashing utility. The embedded controller
> (EC) in these machines often interacts badly with flashing.
> See the manpage and https://flashrom.org/Laptops for details.
> 
> If flash is shared with the EC, erase is guaranteed to brick your laptop
> and write may brick your laptop.
> Read and probe may irritate your EC and cause fan failure, backlight
> failure and sudden poweroff.
> You have been warned.
> ========================================================================
> Aborting.
> Error: Programmer initialization failed.

jcholsap, 2021-04-19:
I think flashrom is telling you that it doesn't like the Winbond Super I/O controller. If that were my laptop, I would proceed to use my RPI, clip it externally, and then see what flashrom thought.

merge, 2021-04-20:
I think this is a duplicate of https://github.com/merge/skulls/issues/173 so if you update to flashrom >= 1.2 things should work. Can you test this?

Let's leave this issue open until we've included a check in our script. thanks.

K0ntrast, 2021-04-22:
**SUCCESS!** 😁👍
I upgraded my flashrom version to v1.2 (as per your advice) which is available through the Debian "sid" repository. This time the internal flashing went through successfully without error. It seems as though I was using flashrom version 1.0 which ships with the latest Debian "stable" branch which produced the error. 

A flashrom version check would be a good thing to add to the skulls.sh script. 

Thank you to both Jay and Martin for your assistance.

commit e5978217 (2021-04-28): functions: add flashrom version check

for >= 1.2

Fixes #191
closed 2021-04-28 by merge