merge/skulls #197: Skulls on T530

swaybar, 2021-06-05:
Would it be possible to build skulls for the T530?

merge, 2021-06-07:
yes, that's possible. I or somebody else who can support it for a long time, just needs the device for testing it before releasing it here. I don't have one. You can fairly simply create a configuration based on the ones we already have for the T430 and build it yourself.

K0ntrast, 2021-11-05:
I will offer to test a Skulls T530 build. :-)

I have recently acquired a ThinkPad T530 that I would like to nominate as a guinea pig for building and testing a Skulls T530 build. I can't imagine this would be too difficult to get working, given that X230 and T430 builds are already available. What would need to be tweaked in the coreboot/seabios build in order to get it working on the T530? Would it really differ much from the T430 build?

K0ntrast, 2021-11-08:
@merge I can send you the BIOS dumps of my T530 if you'd like. I'm more than happy to test it and help support future skulls releases. I plan to use this ThinkPad model as frequently and keep it for the long term.

merge, 2021-11-17:
I got a T530 and will add that soon

tlaurion, 2021-11-19:
@merge some really interesting and practical information was posted at https://github.com/osresearch/heads/issues/996 for T/W 530. Posting it here so that collaboration in bringing up those boards is eased.

prasun-cb, 2021-11-24:
I'm seeing that the PCI-e hotplug config options don't work on T530. It leads to a non booting system. Other than that a standard 4.15 coreboot release works fine.

pgera, 2021-11-24:
I'm seeing that the PCI-e hotplug option leads to a non booting system with coreboot 4.15. Disabling that works fine.

merge, 2021-12-16:
t530 board tested and added

closed 2021-12-16 by merge