merge/skulls #199: Halfway Flashed

Catacombs38, 2021-06-15:
How can I determine if the Bottom Chip was properly Flashed.

How it happened.  I had 1vyrain on my X230.  

I prepared my X230 by rolling back the BIOS to BIOS 2.75 and EC 1.14, and putting back in the Original Intel WiFi chip,  Booted up just fine.  

I updated the Mint Linux 20.1 I have on it.  Ran the bit of program required,  sudo ./skulls.sh -b x230  , computer told me to shutdown, remove battery. and Flash.  Or something like that.

I also removed CMOS battery.   

Computer to Flash from is an 2016 Alienware 17, Core I 7 Skylake, 16 GB RAM.  Ubuntu Hippo, updated, CH341A (Black) with a Pomona Clip.   I attempted Flash with the Alienware Plugged in.   

I had some help to install some Linux pieces, "gcc compiler" and someone explained I needed to provide a file name for the back up file.  Then Installed Flashrom.    You might add those things to Instructions.   

I had difficulty with the Bottom Chip.  So I skipped over that and Flashed the Top Chip.

Bottom Chip Asked what is my programming device.  Said it found some chip, Went off, and then said if it did not find chip automatically it would not Flash, and I should find Chip manually, and run with -c.

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
chip not detected. Please find it manually and rerun with the -c parameter.   

Partially Documented:  https://askubuntu.com/questions/1345849/i-am-still-missing-a-command-somewhere?noredirect=1#comment2299730_1345849
  
As no one came to my aid after a few hours.   I put the computer back together, and it quickly boots showing a splash screen of Skulls with bunny ears.  I set the Mint Linux date/time with Terminal.  Computer gets online.  Seems functional.

When I hit ESC it gives options to:  I am guessing those are the two options as to which Graphics Firmware to use.   

I decided I had done enough weirdness for one day.

I had tried, when it computer was in pieces,, several times to Flash the Bottom Chip.  but I do not see a backup, as I do for the Top chip.  

How can I determine whether the Bottom Chip was properly Flashed?   

If not, then:   Is the instruction, when it says 'find it Manually"  Does this refer to adjusting the Clip.  Or there is another option to show what chips the program thinks it can see?

I have left out some things that occurred, but I think I got everything relevant.    except

I had tried several times to readjust the Pomona Clip.  Now I find one the metal strips that fits into the channel in the Clip for Contact to the Chip is popped out.  As it seemed to program the top chip, I can't guess when that occurred, but I had already failed to program the Bottom one several times.    

I have the low cost Clip that came with the CH341A.  So if I know the path forward, I could try one more time.

This is what I was shown in Terminal.

Please select the hardware flasher: 2
Ok. Connect a CH341A programmer
trying to detect the chip...
Detected MX25L6406E/MX25L6408E.
make: Entering directory '/home/catacombs/Desktop/skulls-1.0.1/util/ifdtool'
make: Leaving directory '/home/catacombs/Desktop/skulls-1.0.1/util/ifdtool'
Intel ME will be cleaned.
The flash ROM will be unlocked.
Start reading 2 times. Please be patient...
flashrom v1.2 on Linux 5.11.0-18-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
catacombs@catacombs-Alienware-17-R4:~/Desktop/skulls-1.0.1$ 

Thanks for any help.

jcholsap, 2021-06-15:
The flashrom utility is telling you that it did not flash the bottom chip. It also saying that it could not automatically determine the details of that chip. Compare to the details it gives for the top chip: Detected MX25L6406E/MX25L6408E. You can find markings physically on the chip to discover it's details. Then you need to research if flashrom supports that chip. It's possible to manually force flashrom to flash using -c parameter to a specific chip. But you'll want to be very very certain you have a backup before trying that.

Catacombs38, 2021-06-15:
thanks for your reply.   

I was incorrect in that by hitting ESC on boot, I get options to boot from the SSD.   
Then  Payload, Nvramui
3. Payload Coreinfo.  
4. Then t. for TPM configuration.  Which leads to current state of TPM is enabled and active, and Ownership has been taken.

I think, what you are saying is, another version of Flashrom Utility would work properly.

This was supposed to be an automatic Process.  So either I did not get the Clip on the Chip properly so it would read.   Or this Chip was not in the group to automatically be Flashed.   MX25L6406E/MX25L6408E. came up when I had the Clip on the bottom Chip.   Might be if I reinstalled everything.  and Only tried to Flash to bottom, it might work?

Strangely, I now see that Ubuntu has an update for Intel Microcode, Processor Microcode firmware for Intel CPU.    

Why Now?   This Processor is very old.  Why an update now?

This evening I have blurry vision from having eyes dilated.   Maybe in a day or two I will get out screwdriver, and see what is on Chip.   

Right now, I hope someone comes along and says, Install this version of FlashRom, and it should work, so Install this version in place of the one I have.  

Thanks for the reply.