merge/skulls #201: 2133mhz Ram Patch

compicat, 2021-09-09:
First of all, thanks for your amazing work and making things easier for us. I just got kingston hyperx 8 gb 2133 mhz ram for my X230. When i check it with memdisk it works 800 mt/s (1600 mhz) as ivybridge had been designed; but i m seeing if we compile coreboot by ourself and patch it devicetree.cb for this; it is possible to get real ram 2133 ram mhz. Can we do this on next release at max_mem_clock_mhz in devicetree.cb? Thanks in advance.

jcholsap, 2021-09-10:
I think this is more of a Coreboot question and how to best initialize the DRAM. If the init process ignores the DRAM SPD data then it could inadvertently overclock the DRAM at startup only to become unstable once operating loading is reached. If the init process trusts the DRAM SPD data, then again there are potential stability issues. (One that immediately comes to mind is Chinese knockoff products.)

I feel it's more complex than just max_mem_clock_mhz = 2133. Coreboot by nature is minimalistic. It may be best to just build it with the settings you want and then test.

One other thought, I think 1vyrain has a menu item fur DRAM speeds. That might be a good starting point.

compicat, 2021-09-12:
@jcholsap thanks for detailed and kind reply.

jcholsap, 2021-09-17:
For further reading on how Coreboot initializes DRAM on Sandybridge controllers and why/how max_mem_clock_mhz bypasses the factory soft fuse limit SCLK 800 MHz (and partly why changing this parameter is your responsibility, not coreboot's):
https://doc.coreboot.org/northbridge/intel/sandybridge/nri_freq.html#devicetree