merge/skulls #203: Using Skulls on Lenovo X230, CH341, Python not present error

Catacombs38, 2021-09-14:
Trying to Install Skulls onto Lenovo X-230 with Ubuntu 20 LTS, CH341 and clip.

When trying to run Skulls on bottom Chip, to eliminate Intel ME, I get message about Python not found.  I have Python 3, and so on.  Three reads successful.   Then at the end.

start unlocking …
./external_install_bottom.sh: ./util/me_cleaner/me_cleaner.py: /usr/bin/python: bad interpreter: No such file or directory

Command to create Python pointer to Python 3 is:

sudo apt-get install python-is-python3

Using Ubuntu 20  LT

Other commands I previously needed to enter.  

https://www.howtoinstall.me/ubuntu/18-04/flashrom/

sudo apt update
sudo apt install flashrom

https://askubuntu.com/questions/192645/make-command-not-found

sudo apt-get install build-essential
sudo apt-get install --reinstall make

When trying to run Skulls on bottom Chip, to eliminate Intel ME, I get message about Python not found.  I have Python 3, and so on.  

Command to create Python pointer to Python 3 is:

sudo apt-get install python-is-python3

At some point I did a massive overall update.    Not sure if that helped me at all.

merge, 2022-11-25:
what's exactly the issue here (if any still)? thank you!

Catacombs38, 2022-11-27:
Sorry,  This represents an effort to Flash Lenovo X230

Someone told me to write up some of the specifics of the help I had received so others would have an easier time.

not really a complaint