merge/skulls #204: How to Flash backup ROM.

Catacombs38, 2021-09-16:
I finished I Flashed the bottom chip with the standard Skulls command, 
Downloads/skulls-1.0.1$ sudo ./external_install_bottom.sh -m -k mom

Seemed to run Normally.  Finished with Flash -

Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.
DONE

Computer will not start.  Only action on pressing power button is light under button.  Plugging into mains does not have light on the back of computer turn on.   

When Skulls runs it shuffles some things around.   So I suppose it unwise to run the same Skulls bottom flash again.   
I should do a Flash where nothing read off the ROM is put back onto ROM.  I presume that can be accomplished if I restore the image saved during the Flash:  

Downloads/skulls-1.0.1$ sudo ./external_install_bottom.sh -m -k mom

and the back up is mom.   

There is no place on the page to explain the command of how to Flash backup ROM.  How to restore to the computer as it was before.  

Thanks for any help

jcholsap, 2021-09-28:
The "flashrom" utility is used to read, verify, and write the SPI EEPROMs. There are several ways, that I can think of, to learn how to use it:
- From the command line, enter "man flashrom" (assuming it's installed)
- Google "how to use flashrom"
- Visit flashrom.org's documentation page

Catacombs38, 2021-10-09:
I have used Flashrom to Flash previous images.   I see the light under the power button when I press it.  Nothing else appears to happen.  

I have one more level to restore top chip back to where it was before installing Core Boot, - alone.   I would also have to change the WiFi inside the computer back to the original Intel one.  

More likely I have damaged some connection from one of the hardware pieces.   Like connection to Keyboard, Touch Pad.   I should also look at screen connection, although I have never touched it.  

The other option is to somehow check through all those pieces and determine if the connections are working.   Kinda beyond my desire to work on that.  

I might try to get another Mobo and install that.   

Or perhaps.   Can I find the original Chip contents from like Lenovo?   Flash that onto the chips.  Someone mentioned a project where the one would find Mobo number.  and install specific chip drivers for heads from that.  

I am tired now.   Some other day.

macandchief, 2023-05-24:
Just for the record I flashed both chips with a Pomona clip (without separate power supply - for tips besides the guide here the yt video How To Actually Libreboot a Thinkpad X200 is helpful) using

`sudo ./external_install_bottom.sh -m -k bottomchipbackup.bin`

and 

`sudo ./external_install_top.sh -b x230 -k topchipbackup.bin`

Everything went fine (Linux booted ok on a dualboot, only Windows did not but I knew about that). Later I decided to use a classic keyboard. For using this keyboard the EC needs to be patched first before flashing Skulls. Therefore I used the backups in order to restore the original state of both chips:

`sudo flashrom -p ch341a_spi -w bottomchipbackup.bin`

and

`sudo flashrop -p ch341a_spi -w topchipbackup.bin -c MX25L3206E/MX25L3208E`

(Thanks goes to the chucknemeth blog for the detailed description there). After restoring the chips I followed the steps described [here](https://github.com/hamishcoleman/thinkpad-ec/blob/master/README.md) and created a bootable usb:

`sudo dd if=patched.x230.img of=/dev/sdc bs=4M status=progress conv=fsync`

Applying the patch went fine and flashed both chips again for using Skulls (it works great, it showes the Skulls splash picture two times described in other threads). As a sidenote if the wiring of the clip is right but the clip is upside down on the chip, I had a normal red light and a weak one on the usb device (otherwise only a normal red one) and received the following message:

```
sudo ./external_install_bottom.sh -m -k bottomchipbackup.bin
Skulls

Please select the hardware you use:
1) Raspberry Pi
2) CH341A
3) Tigard
4) Exit
Please select the hardware flasher: 2
Ok. Connect a CH341A programmer
trying to detect the chip...
chip not detected.
flashrom v1.2 on Linux 5.10.0-18-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Generic flash chip "unknown SPI chip (RDID)" (0 kB, SPI) on ch341a_spi.
===
This flash part has status NOT WORKING for operations: PROBE READ ERASE WRITE
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
No operations were specified.
chip not detected. Please find it manually and rerun with the -c parameter.

```