merge/skulls #212: Skulls 1.0.3 x230 Rom Files Cannot Be Flashed Due to Not Being Able To Be Read

ode79, 2021-11-03:
Neither of the roms from the the latest release can be flashed due to not being able to be read by skulls.sh

```
$ sudo ./skulls.sh -b x230
1) ./x230_coreboot_seabios_free_45ce84113e_top.rom  3) Quit
2) ./x230_coreboot_seabios_45ce84113e_top.rom
file not specified. Please select a file to flash. Please read the README for details about the differences: 1
sha256sum: skulls-1.0.3/x230_coreboot_seabios_free_45ce84113e_top.rom: No such file or directory
skulls-1.0.3/x230_coreboot_seabios_free_45ce84113e_top.rom: FAILED open or read
sha256sum: WARNING: 1 listed file could not be read

$ sudo ./skulls.sh -b x230
1) ./x230_coreboot_seabios_free_45ce84113e_top.rom  3) Quit
2) ./x230_coreboot_seabios_45ce84113e_top.rom
file not specified. Please select a file to flash. Please read the README for details about the differences: 2
sha256sum: skulls-1.0.3/x230_coreboot_seabios_45ce84113e_top.rom: No such file or directory
skulls-1.0.3/x230_coreboot_seabios_45ce84113e_top.rom: FAILED open or read
sha256sum: WARNING: 1 listed file could not be read
```

K0ntrast, 2021-11-11:
I get the same error when trying to internally flash (update) my skulls version.

merge, 2021-11-17:
that's a duplicate of https://github.com/merge/skulls/issues/209

FrostKnight, 2021-11-18:
1.0.2 also has some issues, that would annoy non-expert users...

Aka, I have no idea how often this happens, but at least for me, after it says its loading, the screen doesn't change correctly, but here's the ironic part...  If I type in my password, I login like usual.  As long as I waited more than 10-15 seconds to type it in. 

Bizzare indeed...

darnrain, 2021-11-23:
I put the rom in the same directory as skulls.sh and was able to flash

sudo ./skulls.sh -b x230 -i x230_coreboot_seabios_45ce84113e_top.rom

merge, 2021-12-16:
should be fixed, but as a workaround delete the *.sha256 file.

ode79, 2021-12-16:
Thanks for fixing.
1.0.4 installs without issue.

closed 2021-12-16 by ode79