merge/skulls #215: x230: Blank screen instead of grub

tjburrows, 2022-01-11:
With skulls 1.04 (non-free) and a fresh install of Ubuntu, grub does not display on boot.  Instead, the screen is just blank / black between the Seabios messages and the Ubuntu splash screen.  Ubuntu still boots after the grub timeout, and everything is normal.  I have experienced this phenomena with other linux flavors and other skulls versions.  What could be wrong?  The only thing I can think of is an incorrect nvram setting?

paulmenzel, 2022-07-16:
As you closed the issue, can you please document, how it was solved?

tjburrows, 2022-07-17:
I did not solve this, but it could have been an error on my part since Ubuntu does not always show grub by default.  I have since reflashed stock BIOS, and am no longer having the issue.  I don't use this laptop anymore, so I am closing this issue since I am not able to troubleshoot it anymore.