merge/skulls #220: ThinkPad T440p not seeing Nvidia GPU

mercadolino, 2022-02-19:
Hello, I've installed your coreboot on my T440p, first of all thank you very much for your awesome work, your guide and scripts where very good and I managed to perform the upgrade quite quickly without prior experience.

Now that I have the installation complete I am realizing that once I am in Linux the system is not able to see the dedicated GPU I have on my board, it's an NVIDIA GK208M GeForce GT 730M.

Is that something that has to be added to coreboot?

Thanks for helping out!

nullenvk, 2022-03-13:
I'm not sure how useful this will be, but I am experiencing a similar problem on vanilla Coreboot. The GPU doesn't show up at all in lspci, whereas earlier Coreboot versions used to detect it properly (at least Coreboot 4.13 used to), regardless of whether the GPU's BIOS image was added to the motherboard BIOS. If @mercadolino experiences exact same issues, I think it'd be a good idea to report this to upstream Coreboot's issue tracker.

tlaurion, 2022-03-13:
Have not followed integration of that board and config changes, but have you looked at osboot config files?

nullenvk, 2022-03-13:
@tlaurion Thanks for that. I've done a diff on my config file and osboot's one, and apart from some (probably) insignificant changes related to CBFS size, Azalia settings and some debug options, I haven't noticed any important differences. 

However, I have dumped a Coreboot log using cbmem just now, because I thought it might be useful to diagnose the issue, but I haven't noticed any references to the missing PCI address (which, according to the dumped original GPU VBIOS image, should be `10de:1290`). I'm posting it here too, just in case.
[cbmem_dump.txt](https://github.com/merge/skulls/files/8240621/cbmem_dump.txt)

tlaurion, 2022-03-17:
@nullenvk I may be out of scope totally, but for the work that has been done on the w530-dgpu versions under coreboot, you need to add option ROMs for it to be actually booted and then detected. More blobs, that is. Hope that helps

nullenvk, 2022-03-23:
@tlaurion Thanks for suggestions, but I have already tried out flashing a Coreboot ROM with the dGPU's option ROM built-in, but I didn't observe any differences in comparison with the Coreboot rom without option ROMs. However, I've accidentally stumbled upon this line in the cbmem log:
`[INFO ]  PCI: Static device PCI: 00:01.0 not found, disabling it.`
This is the only line in the log referencing PCI devices not being found, and that address seems to correspond with the one mentioned in this issue:
https://ticket.coreboot.org/issues/310
I don't have much more to say now, but I'll try to investigate this issue further after I finish working on another issue.

nullenvk, 2022-05-01:
@mercadolino @Xanth0k1d @tlaurion
I have found a way to enable the dedicated GPU on T440p, but that required some manual intervention in that model's device tree. I don't have much time to send a proper pull request to Coreboot's Gerrit, but nonetheless, I'd like to provide you with information on how to get it working on your own.

At some point between releases 4.13 and 4.14, some maintainer has decided to disable a PEG device which used to cause problems on the iGPU variant of T440p, assuming that it's unused by the system altogether. However, on the dGPU variant, it's necessary for proper detection of the card. Moreover, I think it may have solved some problems with occasional hangs of that machine's embedded controller, but I'd prefer to do some extensive testing for now. Finally, here's the commit detailing what should you do to fix this issue:

https://github.com/nullenvk/coreboot/commit/aef930dcb6089ef21f00820b2af93654138e1f83

Th3Fanbus, 2022-10-16:
Hi, coreboot developer(s) here. We recently submitted [a tiny change](https://review.coreboot.org/68099) to fix dGPU detection on at least some T440p.

Unconditionally enabling the PEG (PCI Express Graphics) root port at 00:01.1 doesn't hurt. It might increase power consumption a bit, but we can't really do much about it for now. MRC.bin is supposed to initialize PEG ports, but it fails miserably: it will disable the iGPU if it finds a dGPU in a PEG port, and even then it doesn't initialize the PEG port correctly. Fully disabling the PEG ports involves powering down the unused bundles/lanes, and we haven't implemented this yet (no ETA on when it will be done, sorry).