merge/skulls #223: Backlight set to 100% after resuming from suspend to ram

MYEUHD, 2022-03-15:
I just installed Coreboot for the first time on a thinkpad X230 (I installed skulls using 1vyrain)
I tried both the free and the proprietary VGA BIOS roms from the latest release (1.0.4)
And in both cases, after resuming from suspend to ram, the backlight is set to 100%
I'm using X on GNU/Linux, using xbacklight to set the backlight.
After resuming from suspend to ram, xbacklight says that the brightness is set to the maximum
Any help would be appreciated.
Thanks!

darnrain, 2022-03-16:
Put this command into "/etc/default/grub"

“systemd.restore_state=1”

GRUB_CMDLINE_LINUX_DEFAULT="quiet systemd.restore_state=1"

then "sudo update-grub"

reboot

MYEUHD, 2022-03-17:
That didn't work for me.
It's worth noting that my OS isn't using systemd. It's using runit instead.

paulmenzel, 2022-03-17:
Sounds like an OS problem to me? Or does it work with the vendor firmware?

After resuming from STR, are you able to set the backlight level manually or using xbacklight? If yes, you might need to configure your system to run xbacklight on resume, cf. [Screen very dim after resume from suspend](https://unix.stackexchange.com/questions/72948/screen-very-dim-after-resume-from-suspend).

darnrain, 2022-03-17:
> That didn't work for me. It's worth noting that my OS isn't using systemd. It's using runit instead.

Oh yeah I run Arch linux with systemd. But that does sound like a software issue not a bios issue. You could try and download linux mint or ubuntu and make a live usb. Boot the live usb and see if you have the same issue. That should tell you if it's a bios issue or OS issue.

MYEUHD, 2022-03-17:
>Sounds like an OS problem to me? Or does it work with the vendor firmware?

This issue only started to happen after I installed skulls (I was running the stock bios before that)

>After resuming from STR, are you able to set the backlight level manually or using xbacklight

Yes, I can set it using xbacklight. it's just a minor annoyance.

>But that does sound like a software issue not a bios issue.

I thought it would be related to the VGA bios. But the issue persists with both the free software one and the proprietary one.

paulmenzel, 2022-03-17:
Can you please share the coreboot and Linux logs (including when resuming)? Also the Linux logs from the vendor firmware would be good to have.

paulmenzel, 2022-03-17:
In the #coreboot@irc.libera.chat Sean suggested to check the VBT (Video BIOS Table). Maybe compare it between the vendor firmware and the one in the coreboot image.